aredn/aredn #1165: Enhancement - Status of Upload/Download Xfer to node firmware

NickSchmitz, 2024-04-19:
Would like real time progress bar for data transfer of upload/download of firmware to node.

When updating node over remote link/tunnel it can take minutes to xfer data. 
More minutes to program node, re-boot, etc. 

Would like to see FW got there, programming started. Any further details possible.

Current best case you get display of retries after the fact when xfer fails.
But only if data connection NOT reset.

Orv, 2024-04-21:
This appears to have been implemented in the latest nightly build.   Load & recheck, report results.

aanon4, 2024-04-21:
Kind of. We now show progress while flashing, but not while downloading the firmware.

NickSchmitz, 2024-04-21:
Yes, that is the point - I've had problems in data xfer. Upload local image
just starts, top bar goes back & forth for minutes, then connection resets.
No feedback.

I'd like to see feedback data xfer is proceeding correctly. It appears we
now have a timer set for the minutes the flashing / reboot should take
Nb240417.

 When the image it downloaded from the repository, I've seen an error log
stating x%, bytes, errors, etc - dumped after a crash.

Real time feedback pls.

closed 2024-09-05 by aanon4

NickSchmitz, 2024-09-06:
Which build?

20240905-b316964  Doesn’t demo this fix

New UI - Mesh    260    nodes    734    devices

      Old UI - host entries: 260 nodes / 734 total devices 

From: Tim Wilkinson 
Sent: Thursday, September 5, 2024 12:31 PM
To: aredn/aredn 
Cc: NickSchmitz ; Author 
Subject: Re: [aredn/aredn] Enhancement - Status of Upload/Download Xfer to node firmware (Issue #1165)

Closed #1165 as completed.

Closed #1166 as completed.

NickSchmitz, 2024-09-06:
Also NOT implemented with:
firmware version:model: | 20240906-f5ef0ec GL.iNet GL-AR750