aredn/aredn #1168: AR300M still showing [!!!! UNSUPPORTED DEVICE !!!!]

kc2cbd, 2024-04-19:
### Describe the bug
AR300M still showing [!!!! UNSUPPORTED DEVICE !!!!]

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes

### Expected behavior
Shouldn't show anymore

### Additional context
` node: KC2CBD-AR300M-155-CKV
model: GL.iNet GL-AR300M (NAND)

!!!! UNSUPPORTED DEVICE !!!!
boardid: GL.iNet GL-AR300M (NAND)

br-dtdlink E4:95:6E:42:F6:73
br-lan E4:95:6E:42:F6:73
br-wan E4:95:6E:42:F6:73
br0    E4:95:6E:42:F6:73
br0.2  E4:95:6E:42:F6:73
br0.3  E4:95:6E:42:F6:73
br0.4  E4:95:6E:42:F6:73
eth0   E4:95:6E:42:F6:73
eth1   E4:95:6E:42:F6:72
tunl0  00-00-00-00-00-00-00-44-00-00-00-00-00-00-00-00
wlan0  E4:95:6E:42:F6:72

/proc/cpuinfo
system type		: Qualcomm Atheros QCA9533 ver 2 rev 0
machine			: GL.iNet GL-AR300M (NAND)
processor		: 0
cpu model		: MIPS 24Kc V7.4
BogoMIPS		: 432.53
wait instruction	: yes
microsecond timers	: yes
tlb_entries		: 16
extra interrupt vector	: yes
hardware watchpoint	: yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa			: mips1 mips2 mips32r1 mips32r2
ASEs implemented	: mips16
Options implemented	: tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit perf mm_full
shadow register sets	: 1
kscratch registers	: 0
package			: 0
core			: 0
VCED exceptions		: not available
VCEI exceptions		: not available

nvram
hsmmmesh.settings=settings
hsmmmesh.settings.wifimac='e4:95:6e:42:f6:72'
hsmmmesh.settings.mac2='66.246.114'
hsmmmesh.settings.node='KC2CBD-AR300M-155-CKV'
hsmmmesh.settings.dtdmac='66.246.115'
hsmmmesh.settings.config='mesh'
hsmmmesh.settings.nodeupgraded='0'`

aanon4, 2024-04-19:
DIdn't you say this was working last week (see https://github.com/aredn/aredn/issues/1157#issuecomment-2045813166)?

aanon4, 2024-04-20:
Looking for some clarification. Was this issue present after the completion of https://github.com/aredn/aredn/issues/1157 or is this something new?

kc2cbd, 2024-04-21:
Sorry yes it was was showing  [!!!! UNSUPPORTED DEVICE !!!!] after you created the firmware.
It is fixed now. Thank You.

closed 2024-04-21 by aanon4