aredn/aredn #1170: hAP-AC2 all tunnels showing LQ 50%

AI7NC, 2024-04-21:
### Describe the bug
Device has been running 2 days since last reboot. All tunnels indicate 50% LQ in the status page. Neighbor status page indicate 100% quality. MTR to the tunnelled nodes indicate no packet loss and jitter << average ping.

### Interventions tried
Rebooting my device does not resolve the issue. Disable/enable tunnels didn't resolve. 

Running 3.23.12.0 - [ ] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
Tunnels are expected to indicate 100/100 based on packet loss testing.

### Screenshots
Screenshots and support data attached. 

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.
[image]
[image]
[supportdata-AI7NC-QTH-hAP-202404211011.gz](https://github.com/aredn/aredn/files/15052835/supportdata-AI7NC-QTH-hAP-202404211011.gz)

KenHorse, 2024-05-25:
I am seeing similar with the exceptions some tunnels report 100% NLQ while others report either 50% or even 33%. MTR likewise shows no packet loss. I am running 20240516-5069f0e on a hAP AC2

aanon4, 2024-08-18:
Closing as there's been no feedback on this for a while. Reopen if necessary.

closed 2024-08-18 by aanon4