aredn/aredn #1171: TP-link 710 erratic connection/behavior

alchemy33, 2024-04-21:
Situation is a TP-link CPE710 on nightly 20240421 FW.  710 is lan connected via NanoSwitch to a mini-computer running recently updated Linux Mint.  Nothing else on the NanoSwitch.  Connection is via RF to a solo TP-link CPE610 node running FW 3.23.4.0.  The 610 and 710 are close with strong connections.  The 710 has default LQM active initially.

I expect these nodes to indicate reliable connection in Mesh Status and display the alphanumeric node name in Mesh status, WiFi Scan and Chart and to support a simple ping to the numeric address.

I repeatedly observe the behavior of the 710 erratically over several minutes, switches from indicating the 610 by alphanumeric name to numeric IP.  When indicated by numeric, ping will fail.  When numeric is displayed, attempting to browse to the numeric ip will hang with a "unable to connect" browser message.  When numeric IP is displayed LQM with some consistency displays "rf,idle" and NLQ is reliably 0.

When numeric, "rf,idle" is displayed.  When alphanumeric, "rf,pending" or "rf,active" is the LQM condition.

I have observed similar CPE710 behavior with the 4/17 release candidate and this nightly continues the pattern.

The numeric indication appears in Mesh Status, WiFi scan and Chart.  Chart does display a SNR in the 30+ dB range and charts a steady signal even when displaying numeric and there is no apparent functional connection to the 610 (ping fails, browsing fails, NLQ=0)

I observe that over about 30 minutes the status can change between numeric to alphanumeric with a seeming connection to running scan but there seems no consistency.  After 30 minutes it seems to settle a bit and usually numeric (failed connection).

Here is a timeline after a slight change in LQM settings (min SNR=10, min Quality=30) followed by Apply, Save and reboot from setup screen.

1:46 710 boots to Status showing rf evident, Mesh status=alpha & NLQ=100, refresh=same.  WiFi rescan=alpha name and normal, Charting is normal.
1:50 710 Mesh status now numeric (rf,pending) & NLQ=0, refresh = same, refresh now rf,idle, Rescan WiFi=alpha & ok
1:52 710 Mesh=alpha(rf,idle), NLQ=29, SNR=21, Refresh=alpha(rf,active)& NLQ=100, browse to 610 works, ping ok
1:55 610 Mesh=ok, NLQ=100, Browse to 710 works.  This seems expected starting operation so far.
1:56 710 Status shows RF, Mesh=alpha (rf,active) & NLQ=100, Refresh=same, Rescan= OK, Ping = OK
1:58 710 Mesh=alpha(rf,active), NLQ=67, Refresh= same, Chart=OK, Mesh Refresh=OK, Rescan= OK
2:02 710 Mesh=alpha (rf,active)
2:07 710 Mesh=numeric(rf,idle), NLQ=0, Refresh= same, Browse to numeric address fails, Ping fails

This now seems stuck like this.    Checking at 3:25 the CPE710 remains with rf,idle, NLQ=0 and Scan and Chart remain numeric addresses and displaying solid SNR 33-35. Ping fails.

Thanks for looking at this.

aa3jc, 2024-04-22:
I also have the same issues, and even in a test environment, using 2 TP-710’s, set 15 feet apart and pointing directly at each other, they will not completely connect.  Same behavior using a Nano M5 as one point. I have 15 of these that were purchased in Bulk and I find that they are not usable but fully supported by Aredn.  I would like to be in a position to utilize these but right now their worthless except as a paper weight!  If you need assistance with fixing the TP-710, please contact me.

aanon4, 2024-04-22:
The TPLink radios do not get a lot of love I'm afraid. I will take a look at this, but its unlikely to be before June. Sorry.

closed 2025-02-11 by alchemy33