aredn/aredn #1172: Requesting to add two variable fields made available in Aredn table.

aa3jc, 2024-04-22:
Looking to pull two variables from Aredn for the purpose of adding to the table in Meshmap, that includes the devices Serial Number, and the WAN MAC address.  The wireless 'wlan_ip' and 'wifi_mac_address' are already mapped but for Asset Tracking and Placement we would need to track the WAN MAC address and device Serial Number.  This will allow us to have a record of equipment that is purchased by a Grant.  Using both the wan mac address and device Serial Number will qualify as required by the Grant as to where the money was used for.

aanon4, 2024-04-22:
The WAN mac address should be available in the metrics information.
I can provide a serial number, but you'll need to tell me where I can find this.

aa3jc, 2024-04-22:
> [Tracking image][https://github.com/notifications/beacon/ASMLSF5BRR54FOMV6B3F3RLY6VJPNA5CNFSM6AAAAABGTJFH22WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTT3NJJNE.gif]

Tim,

I looked at these two metrics locations:

http://localnode:8080/metrics

http://localnode:8080/cgi-bin/metrics

Neither location shows a WAN MAC Address!
Is there another location I should be looking in? I could just use the wan Mac address for the asset number since it can't be changed or reused.

Most factory firmware has, and displays, the device's Serial Number as verified on the shipping box label.  Once Aredn is loaded it is not available unless the original factory firmware is restored.
This tells me in most cases the serial number is burned in the ROM so it may in most cases be avail in the same method that the Mac addresses are pulled into a variable. 

Thanks for the quick reply and your attention to this issue!

Joe-AA3JC
Joe Caltabiano AA3JC ComL, ComT, Auxcom, ACS
215-237-7991 (cell)
MCAR Mesh:  3103 / 3050 / 3110
PEMA-ACS Mesh: 2002 / 2003 / 2001
HOIP:  100294 (cell)
HOIP:  100299 (pema-acs)
H H: 5702 / 14453
***@***.***
***@***.***
***@***.***

aanon4, 2024-04-22:
There should be a block of network interfaces looking something like this:
```
# TYPE node_network_info gauge
node_network_info{address="d2:3c:6b:2a:d3:c8",broadcast="ff:ff:ff:ff:ff:ff",device="br0.2",duplex="unknown",ifalias="",operstate="up"} 1
node_network_info{address="02:cc:43:a8:a2:6b",broadcast="ff:ff:ff:ff:ff:ff",device="br-wan",duplex="unknown",ifalias="",operstate="up"} 1
node_network_info{address="46:b6:ea:6d:32:9c",broadcast="ff:ff:ff:ff:ff:ff",device="eth1",duplex="unknown",ifalias="",operstate="up"} 1
node_network_info{address="d2:3c:6b:2a:d3:c8",broadcast="ff:ff:ff:ff:ff:ff",device="br0.51",duplex="unknown",ifalias="",operstate="up"} 1
node_network_info{address="00:00:00:00",broadcast="00:00:00:00",device="tunl0",duplex="",ifalias="",operstate="down"} 1
node_network_info{address="d2:3c:6b:2a:d3:c8",broadcast="ff:ff:ff:ff:ff:ff",device="br0.3",duplex="unknown",ifalias="",operstate="up"} 1
node_network_info{address="d2:4c:9f:37:e8:6d",broadcast="ff:ff:ff:ff:ff:ff",device="br0.100",duplex="unknown",ifalias="",operstate="up"} 1
node_network_info{address="00:00:00:00:00:00",broadcast="00:00:00:00:00:00",device="lo",duplex="",ifalias="",operstate="unknown"} 1
node_network_info{address="02:5c:7d:b0:e2:8d",broadcast="ff:ff:ff:ff:ff:ff",device="br-dtdlink",duplex="unknown",ifalias="",operstate="up"} 1
node_network_info{address="d2:3c:6b:2a:d3:c8",broadcast="ff:ff:ff:ff:ff:ff",device="br0",duplex="unknown",ifalias="",operstate="up"} 1
node_network_info{address="02:a6:09:68:ca:27",broadcast="ff:ff:ff:ff:ff:ff",device="br-lan",duplex="unknown",ifalias="",operstate="up"} 1
node_network_info{address="d2:3c:6b:2a:d3:c8",broadcast="ff:ff:ff:ff:ff:ff",device="eth0",duplex="unknown",ifalias="",operstate="up"} 1
node_network_info{address="d2:29:46:a2:1c:e2",broadcast="ff:ff:ff:ff:ff:ff",device="br-nomesh",duplex="unknown",ifalias="",operstate="down"} 1```

aanon4, 2024-12-04:
No feedback received - closing.