aredn/aredn #1174: Grid Square unable to set location

sarcasticadmin, 2024-04-22:
### Describe the bug

- [x] Does this issue still occur on the latest version? (**Required**)

This issue occurs on both `3.23.12.0` and `20240421-d261ebe`
Hardware: `MikroTik RouterBOARD 952Ui-5ac2nD (hAP ac lite)`

If a user filled out only `Grid Square` for `Basic Setup -> Location Settings` it results in an error after hitting `Apply Location Settings`. The `Show Map` button also doesnt display the grid square location. `Upload data to AREDN Servers` fails as well. I dont have the exact error on hand since after putting in Lat and Long I can no longer reproduce the error message.

### Expected behavior
Inputting only `Qrid Square` is enough for `Apply Location Settings` to set location.

aanon4, 2024-04-22:
I tried setting the grid square on a couple of nodes without any issue. You reported getting an error but did not say what the error said - could you provide more detailed information?

aanon4, 2024-11-19:
New UI for this - closing.

closed 2024-11-19 by aanon4