aredn/aredn #1175: Enhancement: provide maximum ETX threshold for routing.

NickSchmitz, 2024-04-23:
**Is your feature request related to a problem? Please describe.**
Many connections are too many hops away to provide any useful bandwidth of connections
to allow listed services to function. Causes excess routing traffic.

**Describe the solution you'd like**
Provide an On/Off switch & ETX threshold number disable node routing.
Off - current routing behavior - all connections routed regardless of ETX
On - ETX less than threshold (4.0 default) are routed.

Default should be Off to minimize impact to casual user.

Discussed with OrvB on zoom call - 4/22

aanon4, 2024-05-15:
So .. do you want to limit what you see on the mesh page? If the ETX > 4 then you're not show the node? Or, as you sort of suggest, do you not want the routing information propagated beyond an ETX >  4. That's a very different thing because if you're node is choosing to do that in some way, then if effects other people's nodes as you're not propagating routing traffic to them - and that seems like a bigger choice. But if you just mean the first option, then there's no effect on routing traffic as this is just a display restriction.

Could you clarify?

NickSchmitz, 2024-05-15:
Hi Tim,

Just limit what is displayed; Don’t mess with real node routing.

Aredn’s mission is to provide links in EMERGENCY conditions.

Who is to say my choice of services (& what ETX) works for that
should limit what someone else sees or uses. 

BTW, I’m seeing worse routing with 3.24.4 (& Nightly 240507, 240513)
than 3.23.12 on bullets & rockets. This may be due to the overall
recent bug fixing – 3.23.12 may let routes thru that don’t fit the 
LQ/NLQ, quality, retries, latency new rules. Just saying, I’d like to
see working routes. 

I will open new issues. 

DO you need a dump of /cgi-bin/sysinfo.json?lqm=1
before & after?

BTW2: I’d like to understand the decision tree of how Aredn
decides a link is routable. What source LUA module contains this?

You should probably help Steve Lewis document the specifics
for others.

Nick

From: Tim Wilkinson  
Sent: Tuesday, May 14, 2024 11:43 PM
To: aredn/aredn 
Cc: NickSchmitz ; Author 
Subject: Re: [aredn/aredn] Enhancement: provide maximum ETX threshold for routing. (Issue #1175)

So .. do you want to limit what you see on the mesh page? If the ETX > 4 then you're not show the node? Or, as you sort of suggest, do you not want the routing information propagated beyond an ETX > 4. That's a very different thing because if you're node is choosing to do that in some way, then if effects other people's nodes as you're not propagating routing traffic to them - and that seems like a bigger choice. But if you just mean the first option, then there's no effect on routing traffic as this is just a display restriction.

Could you clarify?