aredn/aredn #1180: 3.24.4.0 Dangerous Upgrade required for Mikrotik LHG-5nD

k6ccc, 2024-04-25:
When updating a Mikrotik LHG-5nd from Nightly 20240417 to 3.24.4.0 required setting the Dangerous Upgrade switch in order to complete.  Doing so without the switch results in the node reporting the wrong firmware.

Worked fine with the Dangerous Upgrade option set.

k6ccc, 2024-05-12:
Same thing when updating from NB 20240507 to NB 20240512.
I don't remember if that was required for the update from 3.24.4.0 to NB 20240507.

aanon4, 2024-05-15:
What's the node's name? I'd like to take a look.

k6ccc, 2024-05-15:
K6CCC-RB-LHG-5nD
or 10.180.171.159
or 10.165.92.249

This node has needed Dangerous Upgrade for most (not all) upgrades ever since that option was added.  Going further back, when the different Mikrotik files were introduced, I had to use the wrong one quite a few times (sorry, can't remember which one - but it MIGHT have been the 5HPnd).
The node is well connected to the SoCal network.

k6ccc, 2024-05-15:
And here's a support data file for you...
[supportdata-K6CCC-RB-LHG-5nD-202405150715.tar.gz](https://github.com/aredn/aredn/files/15322786/supportdata-K6CCC-RB-LHG-5nD-202405150715.tar.gz)

aanon4, 2024-05-15:
I'm going to have to catch it in the act. The next time this happens and you'd need to flick the dangerous upgrade switch, can you pause and let me know? Thanks.

k6ccc, 2024-05-16:
Updated to NB 20240515 and it took it OK.  I'll try on the next nightly and advise...

k6ccc, 2024-05-27:
Several more updates have gone correctly.  Figures...

k6ccc, 2024-05-27:
Yea, it finally failed again.  Going from nightly 0524 to 0527
Support file attached.
[supportdata-K6CCC-RB-LHG-5nD-202405271455.tar.gz](https://github.com/aredn/aredn/files/15462003/supportdata-K6CCC-RB-LHG-5nD-202405271455.tar.gz)

aanon4, 2024-05-27:
Could you attach a screen shot of the failure (it's not logged) as well as the file you attempted to upgrade it with. And for the moment please dont upgrade.

k6ccc, 2024-05-27:
I was connecting to the LAN IP.  This laptop has a network route via my router - which has an IP from the LHG.
C:\Windows\system32>tracert 10.165.92.249

Tracing route to 10.165.92.249 over a maximum of 30 hops
  1     2 ms     1 ms     1 ms  192.168.203.251  <the router>
  2     4 ms     2 ms     2 ms  10.165.92.249
Trace complete.

The file is:  aredn-20240527-d6a652d-ath79-mikrotik-mikrotik_routerboard-lhg-5nd-squashfs-sysupgrade.bin

Here is the screen capture:
[image]

aanon4, 2024-05-27:
No - I actually want the exact file that failed to upload. Thanks.

k6ccc, 2024-05-27:
It appears that GitHub won't let me upload a .bin file
It's what I downloaded from the AREDN site an hour ago...

aanon4, 2024-05-27:
Try zipping it first. I really need the file you use to see if it's corrupted in some way.

k6ccc, 2024-05-27:
[aredn-20240527-d6a652d-ath79-mikrotik-mikrotik_routerboard-lhg-5nd-squashfs-sysupgrade.zip](https://github.com/aredn/aredn/files/15462285/aredn-20240527-d6a652d-ath79-mikrotik-mikrotik_routerboard-lhg-5nd-squashfs-sysupgrade.zip)

Zipped worked.

aanon4, 2024-05-27:
Thanks. All looks as you'd expect so far.
Could I request temporary access to the device?

k6ccc, 2024-05-27:
Yes, send me a direct email and I'll give you the PW
jim@k6ccc.org

aanon4, 2024-05-28:
So after rebooting the node, I used the file you sent me to upgrade it without a problem via the UI.

k6ccc, 2024-05-28:
I'm coming up with a theory that I will test as new updates come out.  There are three computers that I use for updates, and it appears to only happen when using the laptop from home - and only this one radio.  If I'm on the desktop from home or the Pi-4 which is at home but I VNC into it from work, the updates work.  None of those three computers are directly connected to any AREDN LAN, but rather by way of my router (not consumer grade - each of currently six AREDN nodes connect to the router via different VLANs, and the router gets an IP from each of them).  When connecting from the desktop or the RasPi I can connect to the node name, LAN IP, Mesh IP, or the WAN IP in the case of the two that have a WAN connection.  Because of a DNS issue when connecting from the laptop, I must use one of the IP addresses.

Next update, I am going to try the updates from the laptop and start documenting success or failure and which IP I used for the updates.  See if there is a pattern....

Jim

https://github.com/aredn/aredn/issues/1180#issuecomment-2134397686

k1ky, 2024-07-06:
I have experienced this behavior as well.  No set pattern determined as of yet for conditions as sometimes works... sometimes doesn't.

k1ky, 2024-08-25:
[image]
I have a few nodes that won't upgrade without dangerous upgrade. This is with the latest N-20240825 and before. Coming from N-20240701
[aredn-20240825-7d1d2ad-ath79-mikrotik-mikrotik_routerboard-lhg-5hpnd-squashfs-sysupgrade.zip](https://github.com/user-attachments/files/16741489/aredn-20240825-7d1d2ad-ath79-mikrotik-mikrotik_routerboard-lhg-5hpnd-squashfs-sysupgrade.zip)

k1ky, 2024-12-20:
I don't think I've had any of these problems lately with our units.

closed 2025-07-02 by aanon4