aredn/aredn #1187: Ubiquiti AC devices not scanning if on 10MHz - prior issue 888 not fixed

u-fuzzy, 2024-05-01:
I have been working on troubleshooting this off and on for 2 months.
I just came to report and found the exact issue detailed here in issue 888.
888 is marked fixed but I just updated to 3.24.4.0 which says it fixed the issue but am still experiencing the issue on a LiteBeam AC Gen2.

When I do a Wi-Fi scan nothing appears other than node doing the scan if mesh channel width is set to 10MHz which every node I have seen uses.

Happy to provide more detail if needed but it has already been discussed in [issue 888](https://github.com/aredn/aredn/issues/888) and appears to be the same issue so figured I'd reference that first.

aanon4, 2024-05-01:
Can you describe the steps. Note that the wifi scan now requires you to hit the "Rescan" button once you're on the scan page to generate a scan.

Also, I booted up my LiteBeam AC Gen2 test node, made sure it was on 10MHz, and did a scan. It worked fine (see below):

[image]

u-fuzzy, 2024-05-01:
Interesting that you can scan successfully. Thank you for taking the time to test that.
Unfortunately, I still can't successfully scan as expected.

I can log into the node via ethernet or the second radio configured as a part 15/standard Wi-Fi AP.
I can navigate around the node settings and change things as expected.
From the main page I can go to wifi scan and hit rescan. It takes a moment.
If set to 10MHz channel width it shows only itself.
If channel width is changed to 20MHz I see own and neighboring wifi as expected. I can go back and forth and the results are repeatable.

I have rebooted repeatedly, re-flashed the latest release firmware, and reset to first boot which I assume is the closest thing we have to a factory reset. After initial config and reboot the wifi scan results are again repeatable - good if 20MHZ and nothing but self if 10MHz.
Screenshots below. I'm cleaning one of the support data text files and will add it soon.
Just noticed the nodes MAC changed between 10MHz and 20MHz which seems weird - should be the same radio doing that I would assume... I obscured but showed partially so you can see different MAC in the screenshots even though they're both from the same node.

[image]
[image]

u-fuzzy, 2024-05-01:
Here is txt file from supportdata file.
Removed some info.
All changes I made to this were only removing data and replacing with zeros or a description inside of a pair of two asterisks.
For example, removed part of MAC address or wifi password with the following: **0000** or **WIFIPASSWORD**
[data.txt](https://github.com/aredn/aredn/files/15173992/data.txt)

aanon4, 2024-05-01:
So what other devices on 10MHz would you expect to see?

u-fuzzy, 2024-05-01:
I don't believe it only scans for other devices configured with 10MHz wide channels.
When I change it to 20MHz it see's my home Wi-Fi when it scans, and my home Wi-Fi is using a 40MHz wide channel.
This suggests it is not limited to finding only 10MHz wide networks.

If it only saw other devices using the same channel width, I would at least expect to see a local AREDN node which is using a 10MHz wide channel but I do not see that.

aanon4, 2024-05-01:
Your home wifi might be configured on a channel width of 20 or more, but it beacons at 20 regardless. If you set AREDN to 10 then you will only see other devices at 10. Try this on any other AREDN device and you will see the same thing.

u-fuzzy, 2024-05-01:
Interesting. That doesn't seem right to me coming from a normal Wi-FI world but I don't have any other AREDN devices yet so I can't test unfortunately.

I guess this could be the way AREDN is.
Assuming you're right, I still should be able to see a nearby AREDN node set to 10MHz bandwidth, right?
This node.
[image]

I drove up to it and set a couple hundred feet away pointing this right at it and it saw nothing when set to 10MHz on my node.

u-fuzzy, 2024-07-27:
Sorry for the delay. I can confirm that update 3.24.6.0 has fixed whatever was plaguing me.
Each device I tested had the same issue but is working properly after that update was applied.

closed 2024-07-27 by u-fuzzy