aredn/aredn #1191: On WorldMap, DtD links do not show up

k6ccc, 2024-05-09:
On the worldMap.arednmesh.or web meshmap, DtD links do not show up.  If you click on a node that has DtD link, the DtD links will show on the list, but not on the map.
If you look at this location:
https://worldmap.arednmesh.org/#17.65/34.109573/-117.852522
There should be several DtD links as shown on the edited version file uploaded.

[image]

k6ccc, 2024-05-09:
Listing of one of the nodes that actually has DtD links
[image]

aanon4, 2024-05-09:
This is by design. Short distance DTD links are not shown as it adds unnecessary visual clutter.

closed 2024-05-12 by aanon4