aredn/aredn #1198: hAPac2 on 3.24.4.0 changes 172.31 address on tunnel server on every reboot

KQ6EO, 2024-05-14:
### Describe the bug
A clear and concise description of what the bug is.

After 3.24.4.0 upgrade every reboot causes the tunnel server 172,31.x.x IPs to change and he URL line goes blank.  Tunnels setup with correct IP still works even thought it shows a different IP than when it was created.  You get an invalid URL message when you try to add a tunnel node.

- [ ] Does this issue still occur on the latest version? (**Required**)

Started with 3.24.4.0 release upgrade

### Expected behavior
A clear and concise description of what you expected to happen.

Before the 172.31.x.x address range did not change and the URL did not disappear.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2024-05-14:
Could you add the support data? Issues like this are often the result of some specific configuration corruption.

KQ6EO, 2024-05-15:
[supportdata-KQ6EO-HAP-TSTC-202405132010.tar.gz](https://github.com/aredn/aredn/files/15314739/supportdata-KQ6EO-HAP-TSTC-202405132010.tar.gz)

aanon4, 2024-05-15:
So in this file:

```/etc/config.mesh/vtun```

There should be a line which reads

```config network```

But it's not there. Without it, the node cannot save the ip range and tunnel name. You can log into the node and put it back by hand (make sure there's a blank line before and after it) to fix the problem for now. Then reboot and try setting the tunnels up again.

Not sure why it's missing yet.

aanon4, 2024-05-15:
Following up on this - the `config network` option exists in the base install in the correct file, and there's no code that could remove it. So I'm not sure how it could not be there.

closed 2024-05-25 by aanon4