aredn/aredn #1208: Add antenna to device data

Orv, 2024-05-16:
### Describe the bug
The RocketDish RD-3G26 dish antenna is missing from the Rocket M3 antenna drop down.
- [ ] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
Should appear in dropdown

### Additional context
26 dBi gain, 7 degree beamwidth

ae6xe, 2024-05-16:
Many/All ubiquiti 'M' devices:  add the following antennas -  RD-2G24 RD-3G26 RD-5G30 RD-5G30-LW RD-5G34 AM-9M13 AM-2G15-120 AM-2G16-90 AM-3G18-120:

https://dl.ubnt.com/datasheets/airmaxsector/airMAX_Sector_Antennas_DS.pdf
https://dl.ubnt.com/datasheets/rocketdish/rd_ds_web.pdf

aanon4, 2024-05-16:
@ae6xe failure sure I have most of those - which ones am I missing?

aanon4, 2024-05-16:
If you folk could check that all listed above are either already in the antennas.json file or in this PR - thanks.
https://github.com/aredn/aredn/pull/1209

ae6xe, 2024-05-16:
On ~yesterday's nightly build, the Rocket M3 does not show any of the ubiquity antennas in the list.  The M5 does not show the non-ac airMax rocketdishes, the older models in use (different specs?).    The listings could be tightened up by showing the model #s of these antennas.   See attached pics.
[image]
[image]

closed 2024-05-21 by aanon4