aredn/aredn #1210: Ubiquiti Litebeam AC doesn't populate IP address fields on firstboot

km6lslf, 2024-05-18:
### Describe the bug
A clear and concise description of what the bug is.
Ubiquiti Litebeam AC doesn't populate IP address when AREDN is installed
- [ ] Does this issue still occur on the latest version? (**Required**)
Yes
### Expected behavior
A clear and concise description of what you expected to happen.
Automatic mesh IP address assignment
### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.
[image]

### Additional context
Add any other context about the problem here.

km6lslf, 2024-05-18:
Weird, the screenshot didn't capture the issue. Trying again I get this result when going to localnode:8080
[image]

aanon4, 2024-05-18:
Not quite sure what's going on here, but make sure you wait until you node displayed "NOCALL-<numbers>" before attempting to configure it. It can break if you immediately proceed when it only reads "NOCALL".

aanon4, 2024-11-19:
No feedback - closing