aredn/aredn #1216: on hAP3 running in NAT lan mode, wireguard tunnels are not forwarded in "both"

Marx1, 2024-05-23:
### Describe the bug
A clear and concise description of what the bug is.

- [ ] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2024-05-25:
Can you add the support data? Thanks.

Marx1, 2024-05-25:
[supportdata-W6CX-DATV2-202405251332.tar.gz](https://github.com/aredn/aredn/files/15444947/supportdata-W6CX-DATV2-202405251332.tar.gz)
here you go.

aanon4, 2024-05-25:
The log file is full of this:
```
Sat May 25 13:27:01 2024 kern.info kernel: [177978.331820] br0: port 2(lan2) entered blocking state
Sat May 25 13:27:01 2024 kern.info kernel: [177978.338905] br0: port 2(lan2) entered forwarding state
Sat May 25 13:27:01 2024 daemon.notice netifd: Network device 'lan2' link is up
Sat May 25 13:28:34 2024 daemon.notice netifd: Network device 'lan2' link is down
Sat May 25 13:28:34 2024 kern.info kernel: [178070.891169] qca8k-ipq4019 c000000.switch lan2: Link is Down
Sat May 25 13:28:34 2024 kern.info kernel: [178070.891337] br0: port 2(lan2) entered disabled state
Sat May 25 13:28:37 2024 kern.info kernel: [178074.011788] qca8k-ipq4019 c000000.switch lan2: Link is Up - 1Gbps/Full - flow control rx/tx
Sat May 25 13:28:37 2024 kern.info kernel: [178074.011921] br0: port 2(lan2) entered blocking state
Sat May 25 13:28:37 2024 kern.info kernel: [178074.019351] br0: port 2(lan2) entered forwarding state
Sat May 25 13:28:37 2024 daemon.notice netifd: Network device 'lan2' link is up
Sat May 25 13:30:09 2024 daemon.notice netifd: Network device 'lan2' link is down
Sat May 25 13:30:09 2024 kern.info kernel: [178166.571183] qca8k-ipq4019 c000000.switch lan2: Link is Down
Sat May 25 13:30:09 2024 kern.info kernel: [178166.571353] br0: port 2(lan2) entered disabled state
Sat May 25 13:30:12 2024 kern.info kernel: [178169.691781] qca8k-ipq4019 c000000.switch lan2: Link is Up - 1Gbps/Full - flow control off
```
Any idea what devices in on LAN2 which seem to be constantly rebooting (or something)?

Marx1, 2024-05-25:
Port 2 / Lan 2 is connected to a cisco 3750 -  it's the wan port. The cisco switch shows no logs of the interface going up/down, and it's been up the entire time:

W6CX-NorthPeak-3750x uptime is 1 week, 1 day, 27 minutes

Marx1, 2024-05-25:
I do see PVST Spanning tree enabled. the logs show the bridge going into disabled state - I wonder if the bridge is having an issue with STP? 

I've disabled STP on that vlan, and will pull a new support dump in a bit to see if it's still happening.

Marx1, 2024-05-25:
So I've done some troubleshooting, and I've forced the max speed to 100mb on the port and this is the log:

[180834.092000] qca8k-ipq4019 c000000.switch lan1: Link is Up - 100Mbps/Full - flow control off
[180834.092121] br0: port 1(lan1) entered blocking state
[180834.099555] br0: port 1(lan1) entered forwarding state
[180850.811192] qca8k-ipq4019 c000000.switch lan2: Link is Down
[180850.811366] br0: port 2(lan2) entered disabled state
[180852.891713] qca8k-ipq4019 c000000.switch lan2: Link is Up - 1Gbps/Full - flow control off
[180852.891830] br0: port 2(lan2) entered blocking state
[180852.898914] br0: port 2(lan2) entered forwarding state

this tells me port 2 is actually the DTD port to W6CX-HUB-2

Even though port 2 physically is connected to the switch, and configured as such:

[image]
[image]

On the W6CX-NORTH-HUB2 side:

[image]
[image]
(Mathison KJ6DZB also posted a picture of this in Slack)

Here's the current Support from W6CX-DATV-
[Uploading supportdata-W6CX-DATV2-202405251422.tar.gz…]()

And  W6CX-NORTH-HUB-
[supportdata-W6CX-NORTH-HUB2-202405251423.tar.gz](https://github.com/aredn/aredn/files/15445089/supportdata-W6CX-NORTH-HUB2-202405251423.tar.gz)

North hub is also showing the link flapping. I think this is a secondary issue. As I was seeing that I couldn't get a DTD peering from my house to datv, and if this link is flapping it could be the cause. I did replace the cable with a known good cable.

The main issue is that coming from the Wireguard tunnel,  you can't access the NAT forwards on the device. The camera is physically plugged into the HAP, and I can access it over the dtd link, when it works:

[image]
 but if I have the dtd off, and use the tunnel:
 
 
[image]
 the left is a refresh of the page, and the right is just going to the dns name directly.

Marx1, 2024-05-25:
so I dug more! I did some hard setting of duplex/speeds - I couldn't see it changing duplex anywhere - this is the camera port. I wonder if we have a power/connection problem with the camera; even at 100megabit it still disconnects.

Even still the NAT issue persists.

aanon4, 2024-05-25:
"NAT issue persists" ... I realize, it's just this other stuff makes the logs useless which is a shame.

aanon4, 2024-05-25:
I will note that expecting to override the default management port for a node (80) with something else probably isn't going to work. Can you move the camera to a different port?

Marx1, 2024-05-25:
I'm unable to move the camera - it's located on north peak diablo that requires 4x4 and coordination of multiple people to go up there. 

In terms of "overriding" port 80 - it does work, and has been working on DTD and WAN port. The management moves to port 8080. 

I dumped the firewall from W6CX-DATV2 here :

```

 chain input {
                type filter hook input priority filter; policy accept;
                jump input_lqm comment "block low quality links"
                iifname "lo" accept comment "!fw4: Accept traffic from loopback"
                ct state established,related accept comment "!fw4: Allow inbound established and related flows"
                tcp flags syn / fin,syn,rst,ack jump syn_flood comment "!fw4: Rate limit TCP syn packets"
                iifname "br-lan" jump input_lan comment "!fw4: Handle lan IPv4/IPv6 input traffic"
                iifname "br-wan" jump input_wan comment "!fw4: Handle wan IPv4/IPv6 input traffic"
                iifname "br-nomesh" jump input_wifi comment "!fw4: Handle wifi IPv4/IPv6 input traffic"
                iifname "br-dtdlink" jump input_dtdlink comment "!fw4: Handle dtdlink IPv4/IPv6 input traffic"
                iifname "tun*" jump input_vpn
                iifname "wg*" jump input_vpn
        }

        chain forward {
                type filter hook forward priority filter; policy drop;
                iifname "wg*" jump forward_vpn
                iifname "tun*" jump forward_vpn
                ct state established,related accept comment "!fw4: Allow forwarded established and related flows"
                iifname "br-lan" jump forward_lan comment "!fw4: Handle lan IPv4/IPv6 forward traffic"
                iifname "br-wan" jump forward_wan comment "!fw4: Handle wan IPv4/IPv6 forward traffic"
                iifname "br-nomesh" jump forward_wifi comment "!fw4: Handle wifi IPv4/IPv6 forward traffic"
                iifname "br-dtdlink" jump forward_dtdlink comment "!fw4: Handle dtdlink IPv4/IPv6 forward traffic"
                jump handle_reject
        }

        chain output {
                type filter hook output priority filter; policy accept;
                oifname "lo" accept comment "!fw4: Accept traffic towards loopback"
                ct state established,related accept comment "!fw4: Allow outbound established and related flows"
                oifname "br-lan" jump output_lan comment "!fw4: Handle lan IPv4/IPv6 output traffic"
                oifname "br-wan" jump output_wan comment "!fw4: Handle wan IPv4/IPv6 output traffic"
                oifname "br-nomesh" jump output_wifi comment "!fw4: Handle wifi IPv4/IPv6 output traffic"
                oifname "br-dtdlink" jump output_dtdlink comment "!fw4: Handle dtdlink IPv4/IPv6 output traffic"
                oifname "tun*" jump accept_vpn
                oifname "wg*" jump accept_vpn
        }

        chain prerouting {
                type filter hook prerouting priority filter; policy accept;
                iifname "br-lan" jump helper_lan comment "!fw4: Handle lan IPv4/IPv6 helper assignment"
                iifname "br-dtdlink" jump helper_dtdlink comment "!fw4: Handle dtdlink IPv4/IPv6 helper assignment"
        }

        chain handle_reject {
                reject with tcp reset comment "!fw4: Reject TCP traffic"
                reject comment "!fw4: Reject any other traffic"
        }

        chain syn_flood {
                limit rate 25/second burst 50 packets return comment "!fw4: Accept SYN packets below rate-limit"
                drop comment "!fw4: Drop excess packets"
        }

        chain input_lan {
                ct status dnat accept comment "!fw4: Accept port redirections"
                jump accept_from_lan
        }

        chain output_lan {
                jump accept_to_lan
        }
[... 325 lines cut ...]
All of the rules for the above interfaces have lines that jump to these NAT rules, except there are none for the tunnels and WG:

```
        chain dstnat {
                type nat hook prerouting priority dstnat; policy accept;
                iifname "br-lan" jump dstnat_lan comment "!fw4: Handle lan IPv4/IPv6 dstnat traffic"
                iifname "br-wan" jump dstnat_wan comment "!fw4: Handle wan IPv4/IPv6 dstnat traffic"
                iifname "br-nomesh" jump dstnat_wifi comment "!fw4: Handle wifi IPv4/IPv6 dstnat traffic"
                iifname "br-dtdlink" jump dstnat_dtdlink comment "!fw4: Handle dtdlink IPv4/IPv6 dstnat traffic"
        }

```

WG and Tunnels jump to "accept_vpn" and all that does is accept - it doesn't NAT.

```
        chain accept_vpn {
                oifname "tun*" accept
                iifname "tun*" accept
                oifname "wg*" accept
                iifname "wg*" accept
        }
````

This is probably the real reason it's broken.

aanon4, 2024-05-26:
If you look at the file driving this (/etc/config/firewall) then it does not have the forwarding rules for tunnels. And I suppose this is not a surprise given the UI also doesn't have the option to select forwarding with tunnels (wan, wifi or both). For whatever reason, dtd gets forwarded regardless of the setting.

So what to do? Expanding this selection to a larger number of combinations seems silly and unhelpful. But then "wifi only" is an odd one as dtd always works and wifi here means mesh .. so as long as your request arrives via DTD it'll work even when wifi is not selected?  That doesnt make sense.

We could just add tunnels in here like DTD (always) ... but to me, I'm more inclined to include it when wifi is selected. Thoughts?

Marx1, 2024-05-26:
I agree with the wifi  thought.

My thinking is "wifi" = remote/not local so that would include mesh. DTD  is local, so I'll want it to work.

If we're also going down this thought path - How are xlinks handled firewall wise? because I would want to group this into wifi/remote as well. It's not directly connected - it's across a non-AREDN link. If it's treated as a DTD, it's probably not worth the work to change all the zones etc to split it out as I don't see much benefit besides some corner cases.   I really see it more of Wan  vs  Across AREDN/Tunnel/WG links.

mathisono, 2024-05-26:
There are clearly 3 possible interface types for port forwarding:

-WAN to Mesh LAN
-Mesh to Mesh LAN
-WAN to Mesh

Clearly our problem falls into the "Mesh to Mesh LAN"

aanon4, 2024-05-26:
https://github.com/aredn/aredn/pull/1220

aanon4, 2024-05-26:
So it turned out that this was somewhat complicated because we were kind of hacking in tunnel support on the side of the standard way OpenWRT does things. The PR above should both change that and allow port forwarding over tunnels.

I need someone to test this. It works for the null case, but I dont have any NAT mode devices for proper testing. Volenteers?

Marx1, 2024-05-26:
I can do it. I have a node here at the house that I can have Mathison setup a tunnel for. Let me know what build it's in and I'll update and test it.

aanon4, 2024-05-26:
@Marx1 Let me know the specific device (and preferred email address) and I'll send you a custom build to try. I'd like to do this before it hits nightly.

Marx1, 2024-05-26:
Tested the new code. Confirmed working:

[image]

Able to access via WAN locally, and @mathisono is able to access via the tunnel.

closed 2024-05-28 by aanon4