aredn/aredn #1217: Port mtr to AREDN

kp4djt, 2024-05-25:
At times traceroute just does not provide all of the information MTR does. There is a MTR app for OpenWRT

I would like to be able to install MTR on a node and run an MTR session from that node to gather network information

Traceroute, but it just does not give the amount of information MTR does

Most times I am able to run MTR from a connected machine but there are times when I remote into a node, I would like to gather information from the node but it would imply a trip to the node site and that is not always possible.

aanon4, 2024-05-25:
Is this supported in OpenWRT already?

kp4djt, 2024-05-25:
Here is the URL to the page, looks like it is, I assume that you upload it
to your router:
https://openwrt.org/packages/pkgdata_owrt18_6/mtr

aanon4, 2024-05-26:
Done

closed 2024-05-26 by aanon4