aredn/aredn #1223: HAP AC Lite keeps unadvertising services despite them being up

edkraus, 2024-05-27:
Hello ... this is being discussed on arednmesh.org forum

https://www.arednmesh.org/content/hap-ac-lite-keeps-unadvertising-services-despite-them-being#comment-23407

For me, one HAP will continue to advertise services on a Linux Mint machine running Filegator, Citadel, and a web server for some still photos ... stable.

Added another computer running Windows 10 so I can add Winlink to my local node.  It runs for a while then the node un advertises the service.  If I go back to the admin advertised services page and re save ... and it's back.  The services never went away, just the link.  The computer is pingable.  

I don't know what files you might need so let me know.

Ed

aanon4, 2024-05-27:
As a starting point, can you post the node name so I can take a look? Thanks.

edkraus, 2024-05-27:
Hi Tim!

K7EOK-HAP-ACCESS
let me know if there is any download you need.  the node currently is NOT advertising.  I can download a file then toggle it and do it again so you can compare the state
The Winlink Post Office resides at 10.176.146.68 which has host name EOKW10 

Thanks,Ed

aanon4, 2024-05-27:
Could you attached the support tool data? thanks.

edkraus, 2024-05-27:
here on email or back at the github?  I haven't looked at the github since posting

aanon4, 2024-05-27:
I can ping K7EOK-HAP-ACCESS but I cannot ping 10.176.146.68.

aanon4, 2024-05-27:
Attach it on github if you could.

edkraus, 2024-05-27:
both the ip and the hostname ping from my machine (on same lan)

edkraus, 2024-05-27:
[supportdata-K7EOK-HAP-ACCESS-202405272216.tar.gz](https://github.com/aredn/aredn/files/15462144/supportdata-K7EOK-HAP-ACCESS-202405272216.tar.gz)

Weird, I though I was attaching a file.

edkraus, 2024-05-27:
I can restore the advertising by clicking "Save Changes" even though no changes are made, and regenerate the support data.  Yes?

aanon4, 2024-05-27:
That does is remove the advertising block for a few hours.
Another test - can someone not on your local LAN ping the address successfully?

aanon4, 2024-05-27:
I can telnet to your winlink server, but I cannot ping the address. The node needs to be able to ping the address to verify the device is available. As this is a windows box you might need to adjust its firewall settings.

edkraus, 2024-05-27:
already did that but perhaps not correctly

edkraus, 2024-05-27:
I changed Inbound Rules for File and Printer Sharing (Echo Request) for IPv4 to Yes for Public and Private

aanon4, 2024-05-27:
I think the problem is that you set the winlink server's protocol to `http` and it's very much not that. Setting it to 'winlink' is the way to go.

edkraus, 2024-05-27:
please explain

aanon4, 2024-05-27:
If you setup a service with the protocol http (or https) then we check that the service actually response correctly to http requests (which is a very specific protocol). Winlink is not an http protocol (technically it's telnet, but that a pretty raw protocol) and so will not response to the nodes validation request correctly. If the protocol is not http(s) then we just verify a simple connection is supported.

Why this sort of checking? Because we want people clicking on links marked as http to work correctly in their web browser so they have a good user experience. Winlink marked as http will not do that.

edkraus, 2024-05-27:
OK ... that's why.  I'm asking how.
Do you mean to replace the http in the link with winlink?

aanon4, 2024-05-27:
Oh.
Yes - when setting up the service, in the box you had typed "http" in, instead type "winlink".

edkraus, 2024-05-27:
OK.  I'll try that and then see what happens ... and report back.  Thanks!

edkraus, 2024-05-28:
change made ... will look in a few hours (only 5pm here)
Several questions
1  Does the de-advertising of services that will not answer a ping only apply to http?  Winlink worked fine with the previous firewall rules, the only thing was it wouldn't show up in the AREDN Mesh Node list so users would not be able to find it.  Does this mean I could have left the firewall alone?
2    If I were to offer other services like file server, web server on the windows computer, do I still have to change the firewall rules or AREDN will unlink them if they won't ping?
3    We still haven't solved if my firewall was ok to folks on the local mesh rather than you on the supernode.  I'll get that info later.  Actually, I'll get it myself tomorrow from work as I'll connect with a travel node from there.

Ed

ab7pa, 2024-05-28:
From a documentation standpoint, these are the cases when services will be unadvertised:
[image]
 @aanon4 Tim, let me know if this needs to be expanded to include other details.

edkraus, 2024-05-28:
so it could have been case 3 only all along?

 
 

Screenshot.2024-05-27.at.5.07.12.PM.png (view on web)

AJ6GZ, 2024-05-28:
The Winlink program needs a full URL to be able to find, and correctly, populate its server list.

Mine is:   winlink://AJ6GZ-1-winlink.local.mesh:8772/

So the values are:
Name:  Winlink Post Office   (The program searches for winlink and/or post office)
Link:  [X]  Checked
URL:  winlink 
host  AJ6GZ-1-winlink
Port 8772
Nothing after the trailing ' / '

Yes 'winlink' is a bogus uri, but like Tim said we don't want people clicking on http, telnet, ftp, etc or any of the other things that back in the day would launch external programs in browsers. But you must have some sort of "link" for Winlink to populate hostname and port# correctly.  Otherwise, you'll see it in the list, can add server, but it will fail to work right.

I also allow pings to this host.

You can test by telnet/ssh'ing to your node or better someone else's node further away, then
1) ping the host, preferably by hostname.
2) 'telnet hostname 8772' and see if you get "callsign:" in return. Or whatever your port # is.

If these are true, the service checker and users should be happy.

Re:

" Winlink worked fine with the previous firewall rules, the only thing was it wouldn't show up in the AREDN Mesh Node list so users would not be able to find it.  "

edkraus, 2024-05-28:
Well, the change from http to winlink seems to have worked.  I'll test the ping stuff tomorrow.
Thanks everyone, perhaps a comprehensive checklist for everything you have to do to combine AREDN and Winlink gateways would be good.  We have some instructions for RMS Express, but not as much for RMS Relay.
Ed

edkraus, 2024-05-28:
All working fine, but from another location I cannot ping the server.  Weird.

aanon4, 2024-12-04:
Please reopen if this is still a problem

closed 2024-12-04 by aanon4