aredn/aredn #1224: Refreshing package list reports many bad signatures

KenHorse, 2024-05-28:
Downloading
http://downloads.arednmesh.org/snapshots/targets/ipq40xx/mikrotik/packages/Packa
ges.gz
Updated list of available packages in /var/opkg-lists/aredn_core
Downloading
http://downloads.arednmesh.org/snapshots/targets/ipq40xx/mikrotik/packages/Packa
ges.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/packages/arm_cortex-a7_neon-vfpv4/base/
Packages.gz
Updated list of available packages in /var/opkg-lists/aredn_base
Downloading
http://downloads.arednmesh.org/snapshots/packages/arm_cortex-a7_neon-vfpv4/base/
Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/packages/arm_cortex-a7_neon-vfpv4/aredn
packages/Packages.gz
Updated list of available packages in
/var/opkg-lists/aredn_arednpackages
Downloading
http://downloads.arednmesh.org/snapshots/packages/arm_cortex-a7_neon-vfpv4/aredn
packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/packages/arm_cortex-a7_neon-vfpv4/luci/
Packages.gz
Updated list of available packages in /var/opkg-lists/aredn_luci
Downloading
http://downloads.arednmesh.org/snapshots/packages/arm_cortex-a7_neon-vfpv4/luci/
Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/packages/arm_cortex-a7_neon-vfpv4/packa
ges/Packages.gz
Updated list of available packages in
/var/opkg-lists/aredn_packages
Downloading
http://downloads.arednmesh.org/snapshots/packages/arm_cortex-a7_neon-vfpv4/packa
ges/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/packages/arm_cortex-a7_neon-vfpv4/routi
ng/Packages.gz
Updated list of available packages in
/var/opkg-lists/aredn_routing
Downloading
http://downloads.arednmesh.org/snapshots/packages/arm_cortex-a7_neon-vfpv4/routi
ng/Packages.sig
Signature check failed.
Remove wrong Signature file.
Downloading
http://downloads.arednmesh.org/snapshots/packages/arm_cortex-a7_neon-vfpv4/telep
hony/Packages.gz
Updated list of available packages in
/var/opkg-lists/aredn_telephony
Downloading
http://downloads.arednmesh.org/snapshots/packages/arm_cortex-a7_neon-vfpv4/telep
hony/Packages.sig
Signature check failed.
Remove wrong Signature file.

aanon4, 2024-05-28:
Likely youre running an old nightly build, which means there are no longer packages available for it. Update to the latest nightly.

KenHorse, 2024-05-28:
Running 20240524-108d0da on a hAP AC2 (probably should have included that info before)

aanon4, 2024-05-28:
Not the current nightly.

KenHorse, 2024-05-28:
Ahh, 527. Ok, updated. Don't see a package for mtr however (which is what I was after in the first place)

aanon4, 2024-05-28:
mtr-nojson ... I don't pick the names.

closed 2024-08-18 by aanon4