aredn/aredn #1238: TP-LINK CPE710 NO RF CONNECTIONS IN/OUT POSSIBLE

aa3jc, 2024-06-03:
Once the CPE710 are flashed to Aredn they no longer talk or receive over RF.  I setup a test using 3 flashed CPE710's and aimed one to the other 2 nodes.  All were set to Channel -2 and 10mhz bandwidth.  at a distance of 20ft apart, there was no communications or connections between the 3 nodes.  
Once the CPE710's were flashed back to TP-Link firmware, all three communicated perfectly even with the power turned down.  

This is true with all current builds and nightly builds.

Orv, 2024-06-03:
Joe, suggest adding a support file to this report.

Orv, 2024-06-04:
These other open Issues are for the CPE-710:
#1171
#1150
#731

aa3jc, 2024-06-04:
Attached, please find the latest Support Data for the nightly build #20240604-f2d8adb 
[supportdata-AA3JC-CPE710-HATFLD-000-202406041149.tar.gz](https://github.com/user-attachments/files/15567089/supportdata-AA3JC-CPE710-HATFLD-000-202406041149.tar.gz)
 TP-Link CPE710v.1

ae6xe, 2024-06-04:
not able to read the .gz file.  I'm able to uncompress and tar can list out one file.  Data appears to have been corrupted somewhere in the process

aa3jc, 2024-06-04:
I had placed the cpe710 facing another 5ghz node 2 foot away, and attempted to check out the "Charting" & WiFi Scan functions after which I downloaded the Support Data.  I believe that one of the open tickets addressed this issue where the node gets wonkie after that, and thats when I downloaded the support data.

I rebooted the cpe710 and only downloaded the "Support Data" so I hope you can read this one!

[supportdata-AA3JC-CPE710-HATFLD-000-202406041146.tar.gz](https://github.com/user-attachments/files/15567765/supportdata-AA3JC-CPE710-HATFLD-000-202406041146.tar.gz)

aa3jc, 2024-06-24:
The latest Nightly build 20240618-8d4eade has not changed the status of the nodes "Unable to hear or talk" to another 5ghz node set to the same channel & bandwidth and sitting facing each other 1 foot apart.

apcameron, 2024-06-24:
Please try channel 140 and set it to 20Mhz bandwidth and see if it will communicate then.
If that woks then try again with 10 Mhz Bandwidth and lastly with 5 Mhz Bandwidth

aa3jc, 2024-06-27:
I took your advise above, but re-flashed the node again with the latest Nightly Build, and again it would not communicate.  BUT  after setting the node to channel 140 and 20Mhz bandwidth, on 2 nodes, it is now communicating on at least the channels tested!  Here is a sample of the tests I just completed:
 TP-LINK CPE710 TRANSMIT / RECEIVE TESTS:
Standard for all Tests:
TP-LINK CPE710 v1 
firmware version:  20240618-8d4eade

UBIQUITI LITEBEAM M5
firmware version:  3.22.12.0 

3ft apart, facing each other.
TX Power:  7 dBm
Link Quality Management:  OFF
Distance to Farthest Neighbor:  5.57 miles

-------------------------------------------------------------------------------------
TEST #1:
NODE SETTINGS:
Channel:  140
Channel Width:  20 MHz
OBSERVATIONS:
1.	Litebeam-M5 see’s the CPE710 in WiFi Scan
2.	CPE710 see’s the Litebeam-M5 in WiFi Scan intermittily, freezes, then losses the Browser view and takes over 5 minutes to be accessable again in the browser.

3.	Both Nodes are listed in “Mesh Status” page showing RF connection, only after the CPE710 refreshed or rebooted from the WiFi Scan. Unsure which.
a.	Litebeam-M5 shows the CPE710 with 78.5 TxMbps.
b.	CPE710 shows the Litebeam-M5 with 71.2 TxMbps

TEST #2:
NODE SETTINGS:
Channel:  140
Channel Width:  10 MHz
OBSERVATIONS:
4.	Litebeam-M5 see’s the CPE710 in WiFi Scan
5.	CPE710 see’s the Litebeam-M5 in WiFi Scan intermittily, freezes, then losses the Browser view and takes over 5 minutes to be accessable again in the browser.
6.	Both Nodes are listed in “Mesh Status” page showing RF connection, only after the CPE710 refreshed or rebooted from the WiFi Scan. Unsure which.
a.	Litebeam-M5 shows the CPE710 with 36.1 TxMbps.
b.	CPE710 shows the Litebeam-M5 with 31.2 TxMbps

TEST #3:
NODE SETTINGS:
Channel:  140
Channel Width:  5 MHz
OBSERVATIONS:
7.	Litebeam-M5 see’s the CPE710 in WiFi Scan
8.	CPE710 see’s the Litebeam-M5 in WiFi Scan losses the Browser view and takes over 5 minutes to be accessable again in the browser.
9.	Both Nodes are listed in “Mesh Status” page showing RF connection, only after the CPE710 refreshed or rebooted from the WiFi Scan. Unsure which.
a.	Litebeam-M5 shows the CPE710 with 9.8 TxMbps.
b.	CPE710 shows the Litebeam-M5 with 1.5 TxMbps

TEST #4:
NODE SETTINGS:
Channel:  132
Channel Width:  20 MHz
OBSERVATIONS:
10.	Litebeam-M5 see’s the CPE710 in WiFi Scan
11.	CPE710 see’s the Litebeam-M5 in WiFi Scan losses the Browser view and takes over 5 minutes to be accessable again in the browser.
12.	Both Nodes are listed in “Mesh Status” page showing RF connection, only after the CPE710 refreshed or rebooted from the WiFi Scan. Unsure which.
a.	Litebeam-M5 shows the CPE710 with 15.3 TxMbps.
b.	CPE710 shows the Litebeam-M5 with 21.7 TxMbps

TEST #5:
NODE SETTINGS:
Channel:  132
Channel Width:  10 MHz
OBSERVATIONS:
13.	Litebeam-M5 see’s the CPE710 in WiFi Scan
14.	CPE710 see’s the Litebeam-M5 in WiFi 
15.	Both Nodes are listed in “Mesh Status” page showing RF connection, only after the CPE710 refreshed or rebooted from the WiFi Scan. Unsure which.
a.	Litebeam-M5 shows the CPE710 with 19.5 TxMbps.
b.	CPE710 shows the Litebeam-M5 with 10.8 TxMbps

TEST #6:
NODE SETTINGS:
Channel:  132
Channel Width:  10 MHz
OBSERVATIONS:
16.	Litebeam-M5 see’s the CPE710 in WiFi Scan
17.	CPE710 see’s the Litebeam-M5 in WiFi 
18.	Both Nodes are listed in “Mesh Status” page showing RF connection, only after the CPE710 refreshed or rebooted from the WiFi Scan. Unsure which.
a.	Litebeam-M5 shows the CPE710 with 19.5 TxMbps.
b.	CPE710 shows the Litebeam-M5 with 10.8 TxMbps

TEST #7:
NODE SETTINGS:
Channel:  168
Channel Width:  20 MHz
OBSERVATIONS:
19.	Litebeam-M5 see’s the CPE710 in WiFi Scan
20.	CPE710 see’s the Litebeam-M5 in WiFi 
21.	Both Nodes are listed in “Mesh Status” page showing RF connection, only after the CPE710 refreshed or rebooted from the WiFi Scan. Unsure which.
a.	Litebeam-M5 shows the CPE710 with 65.0 TxMbps.
b.	CPE710 shows the Litebeam-M5 with 72.2 TxMbps
TEST #8:
NODE SETTINGS:
Channel:  168
Channel Width:  10 MHz
OBSERVATIONS:
22.	Litebeam-M5 see’s the CPE710 in WiFi Scan
23.	CPE710 see’s the Litebeam-M5 in WiFi 
24.	Both Nodes are listed in “Mesh Status” page showing RF connection, only after the CPE710 refreshed or rebooted from the WiFi Scan. Unsure which.
a.	Litebeam-M5 shows the CPE710 with 65.0 TxMbps.
b.	CPE710 shows the Litebeam-M5 with 72.2 TxMbps

Overall Observations:
1.	It now appears that the CPE710 IS NOW ABLE TO HEAR AND TALK TO OTHER NODES TRANSMITTING ON THE SAME CHANNEL AND BANDWIDTH.
2.	The CPE710 still acts somewhat wonky, when switching from diffrent screen.  If updating say the channel and width, then clicking the Reboot, instead of getting the Reboot screen with the timer, you see “Hmmm, cant reach this page”.  Sometimes after a reboot the browser does not refresh the page after the node has fully rebooted.
3.	On the CPE710 Channels that are available are 36-64, and 100-184.
4.	On the Lightbeam-M5 Channels that are available are 131-184. So tests will be within these channels only. Not sure if the Litebeam-M5 maybe an International version.

[Chart Scan_aa3jc-litebeam-m5-000-local-mesh-8080-2024-06-27-10_59_04.pdf](https://github.com/user-attachments/files/16016257/Chart.Scan_aa3jc-litebeam-m5-000-local-mesh-8080-2024-06-27-10_59_04.pdf)
[Chart Scan_aa3jc-cpe710-hatfld-000-local-mesh-8080-2024-06-27-10_58_18.pdf](https://github.com/user-attachments/files/16016258/Chart.Scan_aa3jc-cpe710-hatfld-000-local-mesh-8080-2024-06-27-10_58_18.pdf)
[WiFi Scan_aa3jc-litebeam-m5-000-local-mesh-8080-2024-06-27-10_56_47.pdf](https://github.com/user-attachments/files/16016259/WiFi.Scan_aa3jc-litebeam-m5-000-local-mesh-8080-2024-06-27-10_56_47.pdf)
[WiFi Scan_aa3jc-cpe710-hatfld-000-local-mesh-8080-2024-06-27-10_54_48.pdf](https://github.com/user-attachments/files/16016260/WiFi.Scan_aa3jc-cpe710-hatfld-000-local-mesh-8080-2024-06-27-10_54_48.pdf)
l Version.

apcameron, 2024-06-27:
Great. Now are you able to test on channel -2 at 20Mhz, 10Mhz , 5Mhz channel bandwidth.
If that does not work please test all the available channels using 20Mhz.
Then 10Mhz
Then 5Mhz.

If you can provide a spreadsheet of the results it may help in isolating where the problem is.

aa3jc, 2024-06-27:
> [Tracking image][https://github.com/notifications/beacon/ASMLSF3IVCIUH6I4CJUKOWTZJRMK7A5CNFSM6AAAAABIXNUAMSWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTUC3PMZI.gif]

Thanks, will do in the next few days!

Thanks for your help!

Joe-AA3JC

Joe Caltabiano AA3JC ComL, ComT, Auxcom, ACS
PEMA-ACS Assistant ACS Officer- PEMA Eastern Area Office
215-237-7991 (cell)
MCAR Mesh: 215-3103
PEMA-ACS Mesh: 215-2001
HOIP: 100294 (cell)
HOIP: 100299 (pema-acs pbx trunk)
H.H.: 5500000968 (cell)
H.H.: 6100002823 (pema-acs pbx trunk)
***@***.***
***@***.***
***@***.***