aredn/aredn #1244: hAP Lite doesn't properly delete Tunnel client entries

KenHorse, 2024-06-20:
Press DEL button of an existing tunnel client entry appears to be deleted after clicking Save Changes but isn't actually deleted. 

1) Click DEL button
2) Click Save Changes and wait for confirmation
3) Navigate to other weblink (such as Basic Setup)
4) Return to Tunnel Client link and "deleted" entry is still there. 

(a friend noticed this behavior in his hAP)

k1ky, 2024-06-25:
Same issue here on one particular HAP ac Lite - For the past many nightlies, deletion of a Tunnel Client entry comes back after refresh of screen. 
[supportdata-W0ADD-HAP-150-BURNS-202406251849.tar.gz](https://github.com/user-attachments/files/15980498/supportdata-W0ADD-HAP-150-BURNS-202406251849.tar.gz)

aanon4, 2024-06-26:
Look in the file `/etc/config.mesh/vtun` and you should see various lines of data, but the ones which matter begin `config server`. If you find lines without a corresponding name, (eg. `config server 'server_0'` is correct, but `config server` is not) then these lines are broken and must be removed by hand.

I dont know how they end up in these files - we don't write them like that.

k1ky, 2024-06-26:
That got it!  I found a config server section with no 'server x' designation that included information from a previous rename of the node. I deleted that section and all now appears to be stable without bringing back "old" tunnel client entries.  I'm pretty sure this was probably a result of a past node renaming under previous firmware that had some issues with vtun entries.  Probably "NOT" a bug in current firmware, rather some sort of ghost from previous upgrades.

k1ky, 2024-06-27:
Kenhorse - I can probably help you resolve this issue if you haven't already fixed it based on the tips from Tim above.  Feel free to contact me direct k1ky@bellsouth.net and we can get you fixed up and close this issue.

k1ky, 2024-08-01:
The issue originator has NOT responded to my offer for assistance.  Do we have a way to contact him?  If not, I suggest closing this issue as I don't think it's a "bug" rather a result of an upgrade corruption introduced for a prior upgrade.

Rob-HB9XBO, 2024-08-27:
Same problem on a Miktrotik ac3: duplicate tunnel client entry refuses to disappear, and another possibly related problem - just viewing Tunnel SERVER entries causes the Server-Name to be lost and the Network-Adresses to be changed EVERY TIME..

aanon4, 2024-08-27:
Did the solution proposed earlier in this issue not fix your problem?

Rob-HB9XBO, 2024-08-28:
Duplicate client was removed - thanks

KenHorse, 2024-08-31:
Latest result.
hAP is nightly 20240831-638bc9a. It had 2 tunnel clients defined, one legacy and one Wireguard and only the Wiregard is active. Clicked the DEL button next to the inactive legacy tunnel and when the hAP "came back", the inactive legacy tunnel was still listed by the WG one was gone. HOWEVER, the WG tunnel is still active. So as a test, rebooted hAP and no change. WG tunnel still active but not listed and legacy tunnel inactive but listed. This is an AC Lite

aanon4, 2024-08-31:
Want to confirm this is via the new UI?

KenHorse, 2024-08-31:
The new UI doesn't allow for navigation on this particular hAP Lite. In other words, there is no way to get to the tunnels (either server or client) page on the new UI

k1ky, 2024-08-31:
Confirm that you are logged into the unit?

KenHorse, 2024-08-31:
Yup

aanon4, 2024-09-01:
@KenHorse I dont understand your navigation issue. No reason the new UI and hAP Lite shouldnt work together  - they're certainly working fine on the one in front of me right now. Could you post a screen shot of the whole status page? Thanks.

KenHorse, 2024-09-01:
I agree. They work fine on my own hAP AC2. Anyway, screencap attached (i think this is the page you wanted)
[image]

aanon4, 2024-09-01:
You are not logged in. If you were the little figure in the top / right would have a circle round it and the popup menu would say logout rather than login.

aanon4, 2024-09-01:
On the tunnel question itself, did you ever contact @k1ky when he reached out to you in June? He has some experience with faulty tunnel configs. Also, could you attached your support data here so I can also look? Thanks.

KenHorse, 2024-09-01:
Firstly I am logged in (had to use the old GUI to do it however). 
Secondly, I updated my bug report today and made Tom aware of it

KenHorse, 2024-09-01:
Here is the support file from the hAP in question
[supportdata-KH6COM-3-WAILUKU-HAP-EOC-TEST-202408311547.tar.gz](https://github.com/user-attachments/files/16826840/supportdata-KH6COM-3-WAILUKU-HAP-EOC-TEST-202408311547.tar.gz)

KenHorse, 2024-09-01:
I think I see what is going on here.

On my hAP (ac2), I can log in using either the old or new UI and you're logged in both. On my buddy's hAP (Lite), you have to specifically log into the new UI to have access. I originally logged in using the old UI and that gives me full access via the old UI but not the new one. Not until I specifically logged in using the new UI did I have full access using it. Hope this helps

aanon4, 2024-09-01:
> On my hAP (ac2), I can log in using either the old or new UI and you're logged in both.

So no, it doesnt work that way as the mechanisms completely different.

KenHorse, 2024-09-01:
I does on MY hAP ac2

KenHorse, 2024-09-01:
Here it is, just in case
[supportdata-AH6LE-HAP-202408311901.tar.gz](https://github.com/user-attachments/files/16826971/supportdata-AH6LE-HAP-202408311901.tar.gz)

aanon4, 2024-09-01:
Not sure what I can say - I wrote the code - one uses cookies the other uses basic authentication. They are not compatible and neither UI understands anything about the other.

KenHorse, 2024-09-01:
Don't know what I can say either. I'm just reporting what I'm experiencing

K5RA, 2024-09-01:
Experiment just performed on hAP ac lite running latest (20240831) nightly:
1.  Logged in on new UI.  Went to old UI, was not logged in. 
2.  Went to  new UI, logged out.  Went to old UI and logged in (Setup screen).
3.  Went back to new UI, was not logged in.  Logged in to new UI.  Now I am logged in to both. 

I think my results agree with Tim W's observations.  Maybe if you logged out of both new (click logout) and old UIs (reboot?) then log in on new UI and repeat this test.

aanon4, 2024-09-01:
> Clicked the DEL button next to the inactive legacy tunnel and when the hAP "came back", the inactive legacy tunnel was still listed but the WG one was gone.

Just to confirm the steps listed above; did you click the DEL button next to the inactive legacy tunnel, _then hit the Save button_ and then wait for the hAP to come back?

KenHorse, 2024-09-01:
I have figured out the login issue(s). 

Being logged into the old UI does eventually time out after x amount of time and you must log in again. The new UI does not appear to log me out, even after 8 to 10 hours. Hence the confusion about logging into one logs into both. 

As for the original tunnel deletion issue, yes. SAVE was clicked after and the hAP rebooted on its own. When I noticed the wrong tunnel entry appeared to have been deleted, I rebooted again in an attempt to correct it

aanon4, 2024-12-04:
Please reopen if this is still a problem.

closed 2024-12-04 by aanon4