aredn/aredn #1245: AREDN nightly build 20240618 does not resolve named links on mesh status page.  Must use actual IP to access the AR300M16 and any other named links.

convolved-KY4G, 2024-06-21:
### Describe the bug
I downloaded and tested AREDN nightly build 20240618-8d4eade on an AR300M16.  I had to use the hardcoded IP (e.g. http://10.186.171.153:8080/cgi-bin/mesh) to access the node after the upgrade as the named link (e.g. callsign-go-kit.local.mesh) would not resolve.  The same thing happened for all links generated on the mesh status page to include accessing current neighbors and services on neighbor nodes or even the upgraded node itself.  I waited quite a while for the for the names to have time to resolve  with the IPs, but the named links still returned a "ERR_NAME_NOT RESOLVED" page like the one below.
[image]

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] Yes this issue occurs on the latest version.

### Expected behavior
I expected the links displayed on the AREDN page to work as they had before the upgrade.

### Additional context
The links on other AR300M16 node that I had not upgraded work fine to include accessing the node I upgraded to AREDN nightly build 20240618-8d4eade and its underlying services.  That other node is running AREDN 3.24.4.0

aanon4, 2024-06-21:
Could you provide the support data please. Also, I assume this persists across reboots? Can other nodes see this node by name?

aanon4, 2024-06-21:
Could let it run for 30 minutes then do it again? Thanks.

convolved-KY4G, 2024-06-21:
Yes.  I let it run for an hour after it booted and tried again.  It still did not work.  I tried two different browsers.  I still can only get access to the node by typing the IP in directly. Here are some more screenshots of what it shows when I use the text http path "http://ky4g-go-kit.local.mesh:8080/cgi-bin/setup"

[image]
ERR_NAME_NOT_RESOLVED

[image]
 DNS_PROBE_STARTED

[image]
DNS_PROBE_FINISHED_NO_INTERNET

When I type "http://10.14.217.25/cgi-bin/status", which is what http://ky4g-go-kit.local.mesh:8080 should resolve to, I get the following:
[image]

Let me know if there is further testing you need me to do.

aanon4, 2024-06-21:
Sorry. What I meant to ask was, can you resend the support data after the node has been running for at least 30 minutes? Thanks.

convolved-KY4G, 2024-06-21:
OK.  This was after about 45 minutes. Hope this helps.
[supportdata-KY4G-GO-KIT-202406151734.tar.gz](https://github.com/user-attachments/files/15931038/supportdata-KY4G-GO-KIT-202406151734.tar.gz)

aanon4, 2024-06-26:
How big is the network this device is attached to?

convolved-KY4G, 2024-06-26:
It is part of a "go-kit" I built that has the AR-300M16 (the device with the new firmware) connected to a 5-port smart switch.  Normally I have a laptop hooked up to the switch when I need access to it or to upgrade the firmware.  It was connected via RF to another AR-300M16 (running the latest official build, not a nightly release) that is a part of a dedicated WINLINK node which has a separate laptop connected to it.  I believe that was all at the time that was connected.  In fact, I unplugged the device from the 5-port switch and hooked the laptop directly to it to make sure that wasn't part of the problem. Sometimes I have a VOIP phone or another DTD node connected for testing purposes.  I have made IP reservations for these devices based on their MAC.  None of those were connected the day I upgraded the firmware and ran into this problem.

convolved-KY4G, 2024-06-26:
127.0.0.1	localhost
::1		localhost

10.33.219.35	KY4G-GO-KIT	# myself (This is the RF side of the AR-300M16)
10.14.217.28	KY4G	# myself (This is the laptop used for testing the go kit)
10.14.217.26	TL-SG105E	# myself (This is the 5-port smart switch)
10.14.217.27	VOIP-Phone	# myself (This is the voip phone)
10.34.219.35	dtdlink.KY4G-GO-KIT.local.mesh	# myself (This is the same AR-300M16, but for DTD.  I'm not sure why this shows up as a different IP, but the other AR-300M16 appears to do the same thing for itself on the line listed below)
10.24.85.83	dtdlink.KY4G-10-WINLINK.local.mesh	# 10.23.85.83 (See the above comment)
10.186.170.154	VARA-LAPTOP	# 10.23.85.83 (This is the dedicated laptop that runs my WINLINK node that is Packet/VARA-FM/AREDN)
10.23.85.83	KY4G-10-WINLINK	# 10.23.85.83 (This is the RF side of other AR-300M16 that is the only RF AREDN link that was present when I was doing the testing)
10.24.85.83	mid1.KY4G-10-WINLINK	# 10.23.85.83 (mid #1) (I'm not sure why this is here, but it is for the AR-300M16 that has the latest official release that is working fine.)

aanon4, 2024-06-26:
So on very small networks, because there are few OLSR updates, the DNS and OLSR's idea of hostnames can get out of sync. This often happens at startup where the DNS doesnt know anything discovered by OLSR.  On larger networks there sufficient traffic that this problem doesnt happen because things are always being updated. It's an unfortunate race condition without a good solution that doesnt impact general operation.

convolved-KY4G, 2024-06-26:
So is this just unfortunate coincidence that I updated this node with the new firmware and this problem manifested?  Going back to the old firmware would not help I suppose.  A big problem with this is that if I didn't already know the IPs to the other neighboring devices, all of the links to other nodes displayed on the pages are completely useless and appear to remain that way indefinitely.  The node doesn't even appear to be able to resolve its own name to its IP.

aanon4, 2024-06-26:
Well the test to see if this is really what's going on is to log into the node and restart the dns server

`/etc/init.d/dnsmasq restart`

And see if it can then resolve things.

Coincidence? Possibly not - but whenever we update OpenWRT it comes with lots of changes which effects the timing of things.  This is what caused this problem in the past - the restart would be a good test of this. Obviously lots of people are running this code without difficulties, so I'm looking at what's unique to your setup and problem that I'm aware of.

convolved-KY4G, 2024-06-27:
So I logged into the node using PuTTY and restarted the service that you mentioned which did not appear to affect the behavior anything.  I then tried running /etc/init.d/dnsmasq stop followed by /etc/init.d/dnsmasq start (should be the same thing as restart) and it also did not appear to change anything.  I snooped around a bit and found the /etc/hosts file.  I made a /etc/hosts.user file and added some IPs and names to it.  I did the same thing on my Win10 laptop.  This seems to have helped, but seems like a work around instead of DNS taking care of the issue especially for "previously unknown" nodes that join a network.  I'm not sure where to go from here.  I expected restarting DNS would have solved this.  Do you have any other ideas that you want me to try?

convolved-KY4G, 2024-07-03:
I upgraded the node in question from the nightly build to 3.24.6.0 and now I am not experiencing the problem.  I have no idea why the result is different, but it is.  I am going to close this issue since it is no longer being observed.

closed 2024-07-03 by convolved-KY4G