aredn/aredn #1247: AREDN Online Map Data Upload fails - last many nightlies including RC Candidate 20240618

k1ky, 2024-06-26:
### Describe the bug
A clear and concise description of what the bug is
Upload Data to AREDN Servers fails with the following message:  
Configuration NOT saved!
ERROR: Cannot update online map.

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] Yes - for the last several weeks of nightlies including 20240618

### Expected behavior
A clear and concise description of what you expected to happen.
I expect for it to upload successfully and report as such.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.
Sample of the text is listed in the description above.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.
Problem is device independent.  Multiple nodes of different model and manufacture yield the same result.

aanon4, 2024-06-26:
Nothing to do with the AREDN code - the service is currently offline.

k1ky, 2024-06-26:
Copy that - I noted that particular map product is currently unavailable.  The "world map" does work and is picking up and publishing our nodes.

k1ky, 2024-06-26:
Will anxiously await the return of the "static" map. (Any estimate of restoration?) Closing this issue since not related to firmware.

closed 2024-06-26 by k1ky

k1ky, 2024-06-29:
3.24.6.0 Map Data upload working - old original MeshMap is back online and functioning.