aredn/aredn #1249: AREDN Mesh software picker returns wrong file for Rocket XW nodes version 3.24.6.0

k6ccc, 2024-06-29:
When using the software picker on the AREDN Mesh website, selecting Ubiquiti Rocket XW for version 3.24.6.0 results in a file for Ubiquiti Bullet XW.
The node rejects the file as device match failed.

This is the file that gets downloaded:
aredn-3.24.6.0-ath79-generic-ubnt_bullet-m-xw-squashfs-sysupgrade.bin

And here is the error message from the node:
Firmware CANNOT be updated
firmware file is not valid: firmware device match failed

Current Version: 20240618-8d4eade      Hardware Type: (ath79/generic) ubnt (rocket-m-xw)

Note:  This same issue happened with the previous production version.  No problem when using Nightly Builds where you get a list of files to pick (much easier to use).

[image]

aanon4, 2024-06-29:
Do you mean the firmware downloaded from this page?

https://downloads.arednmesh.org/afs/www/?version=3.24.6.0&target=ath79%2Fgeneric&id=ubnt_rocket-m

This downloads the correct file, so I'm not sure what's going on for you unless you mean some other process?

k6ccc, 2024-06-29:
Correct, you just used the page after selecting the type - except the problem is the XW - NOT the XM.

So much easier to have the list of files available like we do for Nightly Builds.

aanon4, 2024-06-29:
Apologies - misread XW as XM.

The one you want is here - https://downloads.arednmesh.org/afs/www/?version=3.24.6.0&target=ath79%2Fgeneric&id=ubnt_rocket-m-xw

Not entirely sure what the double XW is about, except that this isnt new and it first appeared on release 3.23.4.0. In release 3.24.4.0 (the previous one) OpenWRT stuck the Rocket name on the Bullet build as well. We should remove to avoid confusion.

What is the list of files for the Nightly build? If you mean the snapshot directory, then you can always browser the various release files in the same manner. e.g. https://downloads.arednmesh.org/releases/3/24/3.24.6.0/ for the latest.

aanon4, 2024-06-29:
https://github.com/aredn/aredn/pull/1250

k6ccc, 2024-06-30:
"  If you mean the snapshot directory, then you can always browser the various release files in the same manner. e.g. https://downloads.arednmesh.org/releases/3/24/3.24.6.0/ for the latest. "

Yes, that is what I was referring to in the Nightly Builds.  And the link that you provided works fine - provided you know the secret format for the link.  There should be a link to the snapshot directory for the production releases just as there is for the Nightly Builds.  So much faster and easier when you have multiple files to download.  In my case, I currently need to download eight files (was nine until recently) for every release.  Takes me about 30 seconds to download all eight files using the snapshot directories vs about 30 seconds per file using the software picker.

aanon4, 2024-06-30:
No secret formula required - the https://downloads.arednmesh.org/releases/ directory is completely browsable. You should find all the releases dating back to 3.16.2.0 there. And the *really* old stuff is kept in https://downloads.arednmesh.org/firmware/

closed 2024-06-30 by aanon4

ab7pa, 2024-06-30:
There is link on the "picker" page that shows all the downloads available too.
[image]