aredn/aredn #1251: Iperfspeed install not finalizing

kp4djt, 2024-07-10:
Some of my nodes running 3.24.6.0 when attempting to load iperfspeed, take the install but do not show it as installed and registered, so far this has happened on a hAP ac lite and a LHG 5nD. I have installed it on other similar units just fine. I have rebooted the node but it does not correct, I have removed the app (you can see it when you go in and check the list of apps to remove) rebooted the node then re-loaded but get the same issue Same result with the LHG 5nD.

aanon4, 2024-07-10:
(Moved here rather than in docs)

kp4djt, 2024-07-10:
OK, I shall do it. Let me go and remove it from the node that is nearest to me.

kp4djt, 2024-07-10:
I removed the package then after I got removal confirmation I downloaded the support data, then I uploaded the file again,here is what it printed on the screen:
Installing iperfspeed (0.8) to root...
Iperf Speed has been setup at
http://KP4DJT-LHG5-DJT-2-ESN:8080/iperfspeed

Configuring iperfspeed.

Notice the Configuring iperfspeed, it never clears you have to refresh the page or jump to another page and back.
Now 
Here are the two support files the one labeled ..pre.tar is prior to the app load,
post is after

[supportdata-KP4DJT-LHG5-DJT-2-ESN-202407010041-pre.tar.gz](https://github.com/user-attachments/files/16169923/supportdata-KP4DJT-LHG5-DJT-2-ESN-202407010041-pre.tar.gz)
[supportdata-KP4DJT-LHG5-DJT-2-ESN-202407010044-post.tar.gz](https://github.com/user-attachments/files/16169924/supportdata-KP4DJT-LHG5-DJT-2-ESN-202407010044-post.tar.gz)

aanon4, 2024-07-11:
So after the install you dont see it displayed on the node's port forwarding page?

kp4djt, 2024-07-11:
No, nothing. Like it is not there. It does give that URL when it does the
install on the LHG 5
radios, the hap just says it is installing but I get no url. I can use that
URL to reach it, but it
does not publish it. If I try to enter it manually I get the little flag
that says it is not published.

aanon4, 2024-07-11:
If I copy the specific file in /etc/config.mesh which contains the installed info (which looks correct) only my node, the service appears on the ports forwarding page just like you'd expect.

So let me follow up on my second question - what are the differences between this node where it doesnt work compared to a node you have where it does?

kp4djt, 2024-07-11:
Here is the node status screen of a functional node

kp4djt, 2024-07-11:
Here is a node screen of one that does not

kp4djt, 2024-07-11:
Both are the  same type of radio both running the same code, but one
takes it the other takes but does not publish.

aanon4, 2024-07-11:
I didnt get the screen grabs. You need to upload via the website.
So the nodes are identical except for the names?

kp4djt, 2024-07-11:
Yes, I will PM them to you on Slack.

aanon4, 2024-07-11:
Please attach them here so the information is in one place - thanks.

closed 2024-12-04 by aanon4