aredn/aredn #1252: GL.iNET GL-AR300M16-Ext. / new generation / how to flash?

dc4rb, 2024-07-20:
### Describe the bug
I bought a pretty new GL.iNET GL-AR300M16-Ext and found out that this is a new version with
128 MB Memory (see label). It was not possible to flash the latest available firmware due to a checksum error (see error message).

- [ ] Does this issue still occur on the latest version? (**Required**) yes latest version 3.24.6.0 used.
- [ ] older version show the same error
- [ ] the download provides the error message and the label of the new version 
[next_gen_ar300.pdf](https://github.com/user-attachments/files/16319434/next_gen_ar300.pdf)

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
see download

### Additional context
The same firmware in the latest version 3.24.6.0 work perfect on 2 older devices at my site.

aanon4, 2024-07-20:
Could you try this image and let me know if it works - thanks.
https://downloads.arednmesh.org/afs/www/?version=3.24.6.0&target=ath79%2Fnand&id=glinet_gl-ar300m-nand

dc4rb, 2024-07-22:
This image is verified correctly. No checksum error. The router accepts it for upgrade.
After the flashig process the router reappears as original GL.iNET device on 192.168.8.1 not as AREDN on 192.168.1.1
As GL.iNET device everthing seems to work as before. After a second reboot it remains the same.
There is no activity on 192.168.1.1

vy73 Ottmar, DC4RB

dc4rb, 2024-07-22:
not solved yet

tmitchell01, 2025-05-21:
I have the hardware to supply to get this addressed.
Let me know what you need.