aredn/aredn #1253: OLSRD Daemon Crashing, accelerated with higher node counts

AI7NC, 2024-08-01:
### Describe the bug
Running 3.24.6.0 stable on an AC2. This node is the core node for our county. We've seen this issue in 3.24.4 and 3.24.6. We've also attempted a downgrade then re-upgrade. Our node currently has only WG tunnels enabled. We see in our logs that OLSR crashes often, leading to periods of broken connectivity for any traffic that flows through it. The rate of the crashes dramatically increases when we connect to the wider Salem network in Oregon increasing our node count from ~69 to ~360. When connected to the wider network the log shows OLSR crashing pretty consistently every few minutes. When connected to only more local nodes at a count of around 70 the crashes are more intermittent, sometimes only a few times per day or up to 20 per day.

When the event happens, usually all AREDN routing outside the local subnet breaks down, both via tunnels and DTD. Pinging node from a local device usually still works as well as access to public internet destinations via the WAN/internet interface. 

### Expected behavior
OLSR to remain running and stable without breaks often breaks in connectivity.

### Screenshots
Here are some shots of very broken connectivity when we connected to the wider Salem network from our KPI VM. 
[image]
[image]
[image]

### Additional context
We've attached a full support file that includes the log events from an accelerated crash event while connected to the wider Salem network. We can provide logs from periods where weren't connected to the larger network as crashes still occur but not as frequently.

Here's a log summary showing possible correlations between log events while we were connected to the larger network.
[image]

[supportdata-LCcore-Main-202407311821.tar.gz](https://github.com/user-attachments/files/16448685/supportdata-LCcore-Main-202407311821.tar.gz)

aanon4, 2024-08-01:
What's the name of the node at the Salem end?

AI7NC, 2024-08-01:
We have tried 3 separate pathways all  with the same result.

K1RKS-HAP-QTH
W7FLO-GLENADA-RTR
KJ7DZ-WVMN

73,

Nathan Campbell, AI7NC

AI7NC, 2024-08-01:
Here's the log from K1RKS-HAP-QTH that includes the time period where LCCore-Main was connected.

[supportdata-k1rks-hap-qth-202407311643.zip](https://github.com/user-attachments/files/16449096/supportdata-k1rks-hap-qth-202407311643.zip)

aanon4, 2024-08-01:
So I'm struggling to understand what's particular about this node. We have many hap ac2 running with lots of wireguard tunnels and lots of nodes. For example, the supernode W6BI-SOCAL-SUPERNODE is an ac2, OLSRD has crashed twice in the last 10 days (it does crash occasionally), it has 11 wireguard tunnels, and 2309 known nodes. So your node is not stretching the system in any way immediately obvious to me; which is what you might hope for with this issue.

I note that you have LQM turned off. Could you turn it back on? It really shouldnt matter but it's closer to the "default" that way.

Now a more useful thing to try - if you connect to Salem using a non-wireguard tunnel, do you see any difference in stability?

AI7NC, 2024-08-01:
In a previous version we saw LQM disabling tunnels for no reason so we disabled it. I think it may have been related to that quality bug.

We can attempt a legacy tunnel and see if that results in anything different or not. I'll update here on the results of that once we can arrange it.

aanon4, 2024-08-01:
Tunnels really should be at 100% all the time, so if they're not I wonder what's going on with that.

AI7NC, 2024-08-01:
I agree, we ran MTR for a day to the public IP of the node and didn't have any timeouts or significant jitter so we pegged it as a non-internet connection issue.

aanon4, 2024-08-01:
If you look at /tmp/lqm.info, specifically the ping_quality metric, it's not 100 for tunnels and you'd really expect it to be.  This indicates pings are taking > 1 second at times, which is not good. If I compare this to  W6BI-SOCAL-SUPERNODE, every tunnel there is currently at 100.  What kind of Internet is this sitting on?

AI7NC, 2024-08-01:
We noticed all tunnel quality ratings dropped while OLSRD was continually crashing. When we're disconnected from Salem all are 100/100/100 as seen below.
[image]

LCCore-Main is on symmetric gigabit. We're hooked directly into a local ISP as we're in the peering facility. Here's a short MTR from earlier from our PBX VM which sits inside LCCore-Main to the routed IP just outside the firewall for the Salem tunnel server. For some reason they drop pings. 

Salem
[image]

Glenada (Florence)
[image]

edkraus, 2024-08-01:
FWIW

I'm on the opposite end of the WVMN, in the Portland area.  My home node is located on a hAP ac lite via tunnel which used to go thru K9RCP and down to Lane County, then back up to Portland via strange rf paths (which we are working this coming year to improve).  Then I used a tunnel via K1RKS for a while but that also routed into the mid valley and the connection was not reliable.  Currently I am using a tunnel direct to OHSU Sector 1 where the assets I co-manage are located and it's more stable.  I try to avoid the tunnel paths into Salem and Lane County.
 
Since forever (6 months or longer) no matter whom I am tunneled to, my LQ/NLQ always reports 100/50 with total quality 100.  When I view my node from the status page of the node giving me the tunnel it shows me 100/100 and 100.  Go figure.  The sheer number of tunnels in the WVMN is problematic for sure.  

I don't believe the tunnel I'm on has yet been changed to wireguard.  I don't make those decisions for that node.

The rf only assets in the Portland area seem to drop out less often.  I watch statistics collected by instances of meshinfo both locally here in Portland (when they are up) and from further down valley.  

Hope someone can figure this all out.      Ed

aanon4, 2024-08-01:
Morning,

Can you try the following. In the file `/etc/config/olsrd` youll find a line near the top of the file reading `option Pollrate '0.05'` - change this to` option Pollrate '0.01'` then restart olsrd with` /etc/init.d/olsrd restart`

What does this do? Well OLSRD polls for its traffic and if there's a lot it can get behind, kernel buffers fill up and traffic is delayed or lost, so you need to bump up this poll rate. We do this automatically on supernodes because they have lots of traffic on lots of tunnels, but I've never seen it for a non-supernode, but maybe if Salem has a lot of nodes with traffic coming down that one tunnel it can also be a problem.

Note that this change can be overwritten if you change any part of the configuration (not just tunnels) so when you try this and re-enable the Salem tunnel, you'll need to go back in and make the fix again.

aanon4, 2024-08-01:
Something else to consider. One of the things which crashes OLSRD is asking it to give you data via one of its various information dump services. For example, `http://127.0.0.1:9090/routes` will send you a blob of json with all the routing information. Unfortunately these services will often crash OLSRD especially as the internal tables get bigger. I dont know anything about your monitoring, but it's not inconceivable that pulling the metrics information frequency could be causing the problem.

So you might try removing `/usr/local/bin/metrics/olsr.lua` so OLSRD metrics are no longer included.

AI7NC, 2024-08-01:
Hi Tim,

We'll start with dropping all metric pulls from the device to rule that out. We'll monitor via our ping and all-star link metrics for 12 hours to see if it affects the behavior.

edkraus, 2024-08-03:
Well ... the Lane County nodes seem to be back this am, and the Linn County (Albany OR) nodes with ancient software are not appearing now on the mesh.  Is this related or just correlated?

Did Nathan figure out what was causing the issues in Lane County that caused them to isolate their SSID from the rest of the WVMN?  I thought I would read something about this here by now.

Ed

edkraus, 2024-08-03:
And now ... a half hour later I see the Albany OR (LCARES SSID) nodes are back.  What's going on, or not going on?

aanon4, 2024-08-06:
@AI7NC Any news?

edkraus, 2024-08-06:
As of this am, I see three nodes that have LCcore in the node name, and all the ancient nodes in the Albany area are still there.  I have no idea what is going on with this, perhaps Nathan will chime in with an update.  A separate email thread mentioned the hospital and PD in Albany have no interest in upgrading their equipment and some folks are trying to help them limp along with what they have.     I'm not really in the loop just observing what I see.    Ed

edkraus, 2024-08-06:
Looking a little closer, I see some rf nodes in Lane County (Eugene etc) that have a normal SSID.  I don't know if they were changed or not.  Again, someone from that region will need to inform us what is going on.    Ed

edkraus, 2024-08-06:
Don't look behind the curtain .... ooops.  I took a deep dive by going to an installation of meshinfo on a local node and sorted by SSID
We have all on the network
AREDNLCARESFLOnetCPMeshAREDN-BENDAREDN-CHLMsome folks who are long distance tunnel have their mesh SSID (for example I think South Florida)
AREDN-PTP (I'm aware of these)and some with individual ham calls in the SSID
some have -5, -10, and -20 options in the name
I have no idea ... but if someone from the developers group wants to take a deeper dive find an instance of meshinfo on the WVMN and perhaps you'll know if this is ok, or unstable.   Beyond my pay grade .... Ed

AI7NC, 2024-08-07:
Hi Tim,

When we stopped pulling all metrics from our hAP, `lccore-main`, we observed that OLSRD crashes ceased. This was with only a few Lane County tunnels activated; about 70 nodes. Surprisingly, our system remained stable when we expanded tunnel connections to include the statewide AREDN network of 360 nodes. Previously it would crash about 2-3 minutes when connected to the wider network. When we re-enabled metrics one by one, we discovered that the action of pulling the /var/run/hosts_olsr file from the hAP induced OLSRD crashing. One of our metrics related to node count and visibility uses this file for source data.

Basically our metrics management machine, `lccore-hal`, fetched the `/etc/hosts_olsr` file from the hAP via SSH with pre-shared keys once per minute. Each pull involved executing bash command `ssh -p 2222 root@lccore-main.local.mesh cat /var/run/hosts_olsr`. The obtained text file was then parsed on lccore-hal to filter out AREDN devices, retaining only nodes, then posted the metrics to our time-series database. This method was previously stable from 7/4 until 7/13 so it wasn't a correlated change that came to mind when the instability arose on 7/13. 

The ultimate solution was to get away from SSH and use HTTP with the built-in web server. We created a symbolic link in `/www/` pointing to `/var/run/hosts_olsr`. This method has proven stable so far, and we will continue monitoring for ongoing stability.

Nathan, AI7NC

aanon4, 2024-08-07:
Thanks for this update. While I'm glad this has solved the problem, why using ssh and cat to read this file would cause OLSRD to crash is just plain odd.

AI7NC, 2024-08-07:
We agree, we thought nothing of pulling the file once per min via SSH, especially with the VM being a member of the local subnet with a gigabit connection (within limits of the AC2 hardware of course). Why that caused instability while http and a symbolic link does not is beyond me. However, we tried it on/off several times with repeatable instability data so it was a strong confirmation as the cause.

We also tried just a sustained SSH session with the HAP from HAL performing a simple "ls" command every minute for 12 hours with no issues while connected to the wider network. So we don't think it's the SSH session.

aanon4, 2024-08-07:
While crashing your network for science isn't something I think you should do ... I wonder if there would be any difference in your ssh version of this code, if you first copied the host file to a another file, the cat'ed that file? OLSRD updates the hosts file a lot especially on larger networks, so ... maybe ... there's some interaction with ssh+cat and OLSRD when they both use the file at the same time? The copy would be faster than the network cat.

Mostly just trying to understand what this interaction might be. An older version of the code use to copy the hosts file after it was updated, not because of this issue, but in an attempt to keep a stable list available for other services to use - the file often seemed empty when OLSRD was updating it. That never solve the problem though and that code was removed.

aanon4, 2024-08-18:
As no information has been added here in the last 2 weeks, and the problem is fixed if not entirely understood, closing.

closed 2024-08-18 by aanon4

AI7BQ, 2024-08-21:
One working theory relates to the large number of WireGuard tunnels in Oregon that were struggling to startup because some nodes referenced AREDN based NTP servers rather than Internet based NTP servers. Switching everyone to Internet has greatly calmed down the network. Also, switching from SSH CAT to an HTTP mechanism to get the `/var/run/hosts.olsr` file out of the hAP helped. Though considerably more than 30 hours was devoted to root cause analysis, no true cause was found. I agree with the closure of this ticket on this basis. John AI7BQ (LCcore Sysop).