aredn/aredn #1261: Wireguard tunnel does not re-establish if client loses connection (x86 VM under ProxMox)

KenHorse, 2024-08-11:
### Describe the bug
If a Wireguard client loses connection to its server (such as a loss of internet or even a reboot of the client node), the tunnel is not automatically restored. It requires manual intervention on the server side by disabling that client, saving the changes, re-enabling that client and saving the changes again.  This particular server has 2 clients and this issue has been demonstrated with both connections.

Both server and clients are running 3.24.6.0

Connection should automatically be restored without operator intervention being needed.

Attached are support files from both server and client. And from before the restoration of the connection and afterwards. KJ7DZ is server, AH6LE is client

I can provide the config of the ProxMox VM if that's helpful
[After - supportdata-KJ7DZ-WVMN-202408111018.tar.gz](https://github.com/user-attachments/files/16575594/After.-.supportdata-KJ7DZ-WVMN-202408111018.tar.gz)
[AH6LEAfterSupportData.tar.gz](https://github.com/user-attachments/files/16575595/AH6LEAfterSupportData.tar.gz)
[AH6LEBeforeSupport Data.gz](https://github.com/user-attachments/files/16575596/AH6LEBeforeSupport.Data.gz)
[Before - supportdata-KJ7DZ-WVMN-202408111015.tar.gz](https://github.com/user-attachments/files/16575597/Before.-.supportdata-KJ7DZ-WVMN-202408111015.tar.gz)

VA2XJM, 2024-08-11:
Can the nodes access the configured NTP server when not connected to the mesh ???

We had this problem with one of our site using a partner network and without mesh RF link yet. We were indicating a mesh internal NTP server (NTP.local.mesh). Wireguard needs to be time synced to negotiate authentication and then establish the tunnel. Ensure both nodes are using available NTP server when not connected to mesh (mesh internal NTP server can now be signaled inside services, so even with an Internet NTP server indicated, time sync from the mesh is still working).

KenHorse, 2024-08-11:
Yes to both. We run an NTP server on our local mesh

VA2XJM, 2024-08-11:
We do too, but we now let the default NTP server indicated and we add the mesh NTP servers as services as per the docs ( https://docs.arednmesh.org/en/latest/arednGettingStarted/basic_setup.html#optional-settings )

The problem is that while mesh connection is established, the mesh NTP server might not be reachable when NTP try to sync the time. So it will go either in the next hour or in the next 24hrs that it will be reachable. During that time, Wireguard is not able to properly exchange keys to authenticate since time sync is mandatory for it work compared to legacy tunnels.

By setting the NTP server to default `pool.ntp.org` WAN connectivity will be available right away and time will sync. If Internet is not available it will cycle to any published service that will correctly configured:

> If your node is unable to connect to the NTP server specified here then it will briefly search for another NTP service which might be defined on your local mesh network. The node hosting that service must enter its Advertised Service with “NTP” as part of the service name. The protocol should be set to “ntp://”, the hostname should point to the host providing the service, and the port should be set to “123”, the standard NTP port. For example, ntp://ab7pa-box2.local.mesh:123 would identify the NTP server portion of the Advertised Service. See the Configuration Deep Dive section for additional information about Advertised Services.

So reverting to Internet based NTP server + adding services for various mesh based NTP server should fix the problem.

VA2XJM, 2024-08-11:
Yes, both sides will ensure correct time sync.

KJ7DZ is a VM with only WG tunnels, so unless its time is properly set by reachable NTP serber, it will have difficulties to establish tunnels since the calculation is offset due to different clocks between it and the clients that will return invalid key exchanges.

If I understand correctly, AH6LE seems to be a physical device with radio that may have mesh access once booted and may also take some time to sync its clock if the NTP server is not local (waiting for routes to propagate).

KenHorse, 2024-08-11:
That did indeed fix our problem. While not a bug, perhaps this needs to be emphasized in the docs? Also, I cannot seem to find anything at your link regarding more than one NTP server definition

VA2XJM, 2024-08-11:
Yes. I was not able to test the fix in a lab environment since my lab is offline due to heavy works at home and no one reported it before, so I tough that maybe I was the only one with this kind of setup. 

However you confirm the fix is right. I'll make a PR to the doc repo.

I saw you edited your post, but for the records and maybe helping someone else : Each NTP server must be published as a service on it's local node.

AI7NC, 2024-08-11:
What's the time sync threshold for successful key exhange? Within 60 seconds? within 1?

VA2XJM, 2024-08-11:
> What's the time sync threshold for successful key exhange? Within 60 seconds? within 1?

IIRC it is within 5minutes.

KenHorse, 2024-08-11:
So the connection isn't persistent? Both node and client would have a set clock if neither was rebooted, yes?

VA2XJM, 2024-08-11:
> So the connection isn't persistent? Both node and client would have a set clock if neither was rebooted, yes?

AI7NC was questionning about the allowed offset of clocks. So you do not have to be in total sync. There can be a slight offset between the client and server clock. Past a certain threshold it will fail.

The Wireguard design is made so a handshake happens every 2 minutes. During the handshake, encryption keys are refreshed for the next 2mins. To generate the encryption key at every handshake, both clocks needs to be in relative time sync and have the correct predefined key. If the handshake fails, the tunnel will not work anymore.

The tunnel is not a persistant TCP tunnel, it is `stateless` and using UDP. So both sides listen on their UDP port for incoming traffic and decrypt on receive. This means that if there is no traffic  sent through the tunnels, there is no packet exchanges until the next handshake.

VA2XJM, 2024-08-11:
Doc PR done: aredn/documentation#379

Fixed. @KenHorse feel free to close the issue :)

KenHorse, 2024-08-11:
So if clocks on both server and tunnel were within 5 minutes of each other, why won't they automatically re-connect if the issue was a temp internet connection that resolved itself? Just curious as to the mechanism

VA2XJM, 2024-08-11:
The 5 minutes mark is what I have on top of my head, but it can be less than that... Let's assume I am wrong it is 30 seconds for a quick example.

So both nodes are "in sync" with an offset of 15seconds. One of the system (client or server) is under intense load and keep its head out of the water by resyncing with the NTP server every hours, but at one point Internet drop and the VM can't re-sync, the offset will get greater and greater. When Internet is back, it can't reestablish the tunnels due to increased offset that passed over the threshold of WireGuard. So it do not have access to the mesh based NTP server to re-sync because the tunnel fail at authentication.

In that case, if you offset of 15 seconds per hour, you miss one hourly sync, you are now 30secs offset at the next hour, so the tunnel drop and no time sync is possible using mesh based NTP server. In case of a reboot, you reboot at unix time or more or less random time until NTP can sync.

With a VM reboot, it is possible that Proxmox pass the unix time value so the clock is in sync at boot will then drift slowly. (I am not yet using ProxMox, it is in my to do list to migrate VBox to ProxMox but VBox have a setting to send UNIX time to hte VM at boot)

The legacy tunnel were not time dependent, so I presume that just like me, you migrated on WireGuard and it worked for a while until one time you had troubles and tried many things that made it work again until the next time...

AI7NC, 2024-08-11:
Yeah, drift will occur over time. This can be made worse if you aren't syncing at the VM level and the hypervisor isn't tied to an NTP server.

KenHorse, 2024-08-11:
I know at least 2 instances where neither the client nor server actually rebooted (or otherwise restarted). In this example, the client lost internet connectivity for an unspecified period of time. When service was restored, the client wasn't able to re-establish the WG tunnel and the procedure I explained in the OP was necessary to get them connected again. 

Thanks for explaining the mechanism. It's been helpful to understand. I'll close this for now, thanks

closed 2024-08-11 by KenHorse