aredn/aredn #1262: Flashing factory.bin Fails on Ubiquiti Nanobeam 5AC WA

AI7BQ, 2024-08-13:
### Describe the bug
Flashing factory.bin to Ubiquiti Nanobeam 5AC (WA) fails while executing:

```
/tmp/fwupdate.real -m /tmp/factory.bin
```

I receive the following error message:

```
Invalid version 'WA.ar934x.v8.5.3-42.AREDN-r23809-234f1a2efa'
```

- [x] Does this issue still occur on the latest version? (**Required**)
Yes.

### Expected behavior
I expected the flashing process to complete and the device to reboot as per the last action item in _Step 3. Install the firmware_ in the section titled _Ubiquiti 802.11ac First Install Process_ in the _Installing AREDN Firmware_ document.

### Screenshots
N/A

### Additional context
I am following all instructions in the **Installing AREDN Firmware** document. I am qualified as an embedded computer engineer and I have experience flashing well over a dozen AREDN devices. However, this is the first time I have flashed this model, so I have no reference. I do have a second identical Nanobeam which I have not tried to flash.

I successfully downgraded the Nanobeam 5AC to:

```
WA.v8.7.4.45112.210415.1103.bin
```

I am using the following AREDN "factory build" binary:

```
MODEL:    Ubiquiti NanoBeam AC Gen2
TARGET:   ath79/generic
VERSION:  3.24.6.0 (r23809-234f1a2efa)
DATE:     2024-06-28 17:28:34
FILE:     aredn-3.24.6.0-ath79-generic-ubnt_nanobeam-ac-gen2-squashfs-factory.bin
```

southmookie, 2024-08-20:
I purchased two NanoBeam AC 5 Gen 2 (WA) devices for a high-profile hospital installation, and I need to get this working before 9/13/2024 at the latest. Please don't let this bug get sidelined because of the flurry of UI Github tickets.

AI7BQ, 2024-08-20:
I purchased two NanoBeam AC 5 Gen 2 (WA) devices for a high-profile hospital installation, and I need to get this working before 9/13/2024 at the latest. For what it's worth, the md5sum of /bin/fwupdate.real is not 3584df67dd90aeedf4bee71c08292233

aanon4, 2024-08-20:
So reading through the notes quickly, it says you downgraded to `WA.v8.7.4.45112.210415.1103.bin`. Does this mean you were unable to downgrade to `WA.v8.7.0.42152.200203.1256.bin` on this NanoBeam AC Gen2?

AI7BQ, 2024-08-21:
Yes. You are correct. I have been unable to downgrade both of my NanoBeam AC 5 Gen 2 (WA) devices to firmware version WA.v8.7.0.42152.200203.1256.bin

I have tried about six times. Each time, when I do this.:

`Systems page |  Firmware Section | Upload Firmware | Select WA.v8.7.0.42152.200203.1256.bin | Click OPEN button`

I receive the following feedback.:

`Uploading... Checking uploaded file... ! Warning: Incorrect firmware version. Please upload v8.7.4 or newer and try again. `

Brett KG7GDB and I repeated the procedure in lock-step unison without shortcuts or assumptions (I have the transcript), and got the same results each time. His comment was _"That's an issue I have not seen and apparently neither have [the] developers."_

aanon4, 2024-08-21:
So this issue has come up with some Litebeam ACs which cannot be downgraded to 8.7.0.  As you've noted, you downgrade your Nanobeam AC to 8.7.4 but that doesnt work because there's also a change you need in the firmware. I'll submit a patch shortly.

aanon4, 2024-08-21:
https://github.com/aredn/aredn/pull/1353

aanon4, 2024-08-21:
Please try build tomorrow. Make sure you've downgraded your Nanobeam AC to 8.7.4

aanon4, 2024-08-22:
The current nightly build has the change. Let me know if it works.

aanon4, 2024-08-22:
@dan97520a This was not a general fix, but a fix for a specific device type. Please file a separate issue for the PowerBeam so we can track that.

AI7BQ, 2024-08-23:
Issue Status: FIXED
Next Step: OK TO CLOSE
Summary: The changes implemented in the 0240822 nightly build resolved the problem as originally defined. Thank you Tim.

closed 2024-08-23 by aanon4