aredn/aredn #1264: Latest Nightly Build on hAP AC2 not showing new GUI

KenHorse, 2024-08-16:
### Describe the bug
Using FireFox, cleared the cache but don't see any difference in GUI

- [ ] Does this issue still occur on the latest version? (**Required**)
20240816-0432bf3 

### Expected behavior
Expected new GUI ala YT video

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2024-08-16:
Could you add a screen shot of the node's status? Thanks.

KenHorse, 2024-08-16:
Argghh, I just noticed the new button to select the new GUI. Sorry, it's good

closed 2024-08-16 by aanon4