aredn/aredn #1265: AREDN Logo Click Function now redirects to AREDNMesh Web site instead of home node

k1ky, 2024-08-16:
**Is your feature request related to a problem? Please describe.**
The function of the AREDN Logo (upper left corner of new GUI screen) selection has changed.  
**Describe the solution you'd like**
Please revert function to select HOME node and maybe add new Icon to redirect to AREDNMesh Web site.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.
Add new ICON for redirect to AREDNMesh Web site, Keep old functionality on the old logo.

**Additional context**
Add any other context or screenshots about the feature request here.

aanon4, 2024-08-16:
Works as intended, but I'm open to discussion about whether the old functionality was preferred by people; but I dont think an extra icon just to get to the arednmesh website is warranted.

k1ky, 2024-08-16:
My 2 cents - I think the upgrade should be as transparent as possible regarding how old "features" worked.  This is clearly a "change" in behavior and I really like the ability to quickly revert to my "home" node when browsing around as before,  I have favorites in my browser to quickly go to the AREDNMesh Web site.

ab7pa, 2024-08-16:
On the old UI clicking that logo would take you back to the node status page, but with the new UI you are already there -- there's no need to have an icon that takes you back to node status.  Makes sense to me that it would navigate to arednmesh.org if it is available.  Or we could remove the link altogether if it's unnecessary.

aanon4, 2024-08-16:
I think Tom's argument is that regardless of which node you are on anywhere on the mesh, the AREDN logo will always take you back to your localnode.  That's not the case in the new UI. It's not functionality I personally use (I have bookmarks for that).

k1ky, 2024-08-16:
I have a bookmark for the AREDNMesh Web site in my browser.  I guess I could add a Bookmark in my browser to revert back to the home node. I use the old feature a lot.  Will be interesting to see what the community has to say (if they even knew that feature existed!) I believe this original feature request was something that I submitted long ago and you creatively made it available by clicking on the Logo, which I thought was a novel solution!

aanon4, 2024-08-16:
One option would be put the original functionality back on the icon, and add the AREDN website link into the top/right menu   with the Help link.

k1ky, 2024-08-16:
Works for me!

aanon4, 2024-08-17:
Done

closed 2024-08-17 by aanon4

mathisono, 2024-08-18:
I already mistakenly clicked on the icon and was taken back web site....

Im voting for the original functionality back on the icon!