aredn/aredn #1266: No Commit function available when deleting legacy tunnels.

Orv, 2024-08-16:
### Describe the bug
On a tunnel server with many tunnels, when deleting disabled legacy tunnels Commit button doesn't appear,

- [ ] Does this issue still occur on the latest version? (**Required**)  Yes, new UI.

### Expected behavior
This is the Ventura County tunnel server, an ac2 with the largest known number of tunnels on the AREDN network.
Note - this function worked with my home ac2, which only has a few tunnels
[supportdata-K6PVR-VC-TUNNEL-SRV-2024-8-16-11-26.tar.gz](https://github.com/user-attachments/files/16640495/supportdata-K6PVR-VC-TUNNEL-SRV-2024-8-16-11-26.tar.gz)

aanon4, 2024-08-16:
I'm assuming they dont actually get deleted?

KD4WLE, 2024-08-17:
I'm seeing the same with Wireguard tunnels on a Proxmox node, There is no commit button after removing the tunnel.

aanon4, 2024-08-17:
Same question I asked above - do the tunnel actually get deleted?

aanon4, 2024-08-17:
Also, as I can't reproduce this, I'm probably going to need access to a node with this problem.

KD4WLE, 2024-08-17:
Hit me up off list, and ill let you lose in kd4wle-titusville-tunnelserv.local.mesh

***@***.***

aanon4, 2024-08-17:
Apparently Javascript in the browser and Javascript in OpenWRT disagree on exactly how to decode a url encoded string. Not sure exactly how yet.

aanon4, 2024-08-17:
... sort of. It seems url encoded arguments are limited to 4k in length, so the problem here is that there are lots of tunnels and so the argument gets too long. I will split it up to avoid this issue.

Orv, 2024-08-17:
Now works as expected (at least deleting one tunnel)

closed 2024-08-17 by Orv