aredn/aredn #1270: Featur Request - Firmware Revision list

KD4WLE, 2024-08-16:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
A list of local nodes, and their Firmware revision level for those of us that manage a large number of nodes. 

**Describe alternatives you've considered**
Currently I log into each node to check revisions. it's time consuming, and often find I have missed updates on some nodes. 

**Additional context**
Currently using 3rd party software for this, but would be nice if it was built in somewhere.

k1ky, 2024-08-25:
We host a version of KG6WXC's MeshMap on a few servers on our Soutmesh system.  Here is a sample screen that shows firmware version number on each node in the system.  You can adjust the poll frequency.
[image]

aanon4, 2024-12-04:
No plans to add cluster management to a node. This should be done as a separate program or package and not built into the AREDN base image.

closed 2024-12-04 by aanon4