aredn/aredn #1271: Add ability to copy & paste from new UI pages

Orv, 2024-08-16:
**Describe the solution you'd like**
Add ability to copy & paste from new UI pages.    It works from the old UI; would be nice to continue to be able to do it.

.

aanon4, 2024-08-16:
Anything in particular?

KenHorse, 2024-08-16:
FWIW, I C&P IPs and MACs all the time

k6ccc, 2024-08-16:
For example.  After updating all my nodes to the latest nightly, I posted this note on MatterMost:
All K6CCC nodes updated to Nightly build 20240816-0432bf3 
Of course I typed the first part, but the build number, I would normally copy the numbers from one of the nodes status page rather than typing it (avoids errors).  Could not highlight and copy the text on the status page.
And as with KenHorse, I C&P IP addresses into other stuff somewhat routinely.  Now if you are logged onto the node and bring up the edit page, you can highlight and copy from there.

k6ccc, 2024-08-17:
Partially fixed in 20240817.  On fields that bring up a popup window to make changes (such as the radio information, firmware information), it does not work because when you highlight what you want to copy, and then release the mouse button, it brings up the popup window.  When you close the popup window, your selection is no longer highlighted.  On fields that do not bring up a popup window, copy works fine.

aanon4, 2024-08-17:
That should fix the popups.

aanon4, 2024-08-18:
Let me know how this looks today.

k6ccc, 2024-08-18:
Yep.  Working today on the one node I have updated so far - a hAP lite

Thank you!

closed 2024-08-18 by aanon4