aredn/aredn #1276: new UI

mathisono, 2024-08-16:
Note: wifi signal tool page docent work / display data in when in wan wifi client mode.

Version Nighty 

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2024-08-17:
What does the current UI do for this case?

mathisono, 2024-08-17:
In the old UI, im unable to find the wifi scan. It appears to be hidden. The old UI dose show the following: 
[image]

mathisono, 2024-08-17:
I dont see a cell with WAN info, consider adding one for wan info.

aanon4, 2024-08-17:
Done

closed 2024-08-17 by aanon4