aredn/aredn #1277: New UI : To Do "prompt" docent produce

mathisono, 2024-08-16:
New UI : To Do "prompt" docent produce a solution. 

Im would expect the To Do prompt , to prompt a user with a link , linking to the appropriate page for this setting, or an option to dismiss the prompt all together.

[image]

mathisono, 2024-08-17:
Sugest placing a persistent prompt in the location cell and not at the top UI page.

fmagic01, 2024-08-23:
Agreed with the original suggestion--it would be handy to be able to dismiss the todo window. For example, my node is used for portable operation and I will not be setting a location on it because that will change regularly. I want the todo window to not appear in this case.

aanon4, 2024-08-23:
Whiie I dont want to make these notices discreet (which is against the reason for them) I do think we need a mechanism to dismiss any (not just these) messages. My question then is, when you dismiss a notice without addressing it, when should it return? I dont think never is a good option.