aredn/aredn #1284: Add Interenational Standard Units (metric)

kp4djt, 2024-08-17:
**Is your feature request related to a problem? Please describe.**
I see distance in miles, but we have international folks now using AREDN, and some of us just prefer metric

**Describe the solution you'd like**
Soft switch to select US Standard Units or International Standard Units

**Describe alternatives you've considered**
Would have to make conversion external to node data

**Additional context**
Not a show stopper but except for the US, Liberia and Myanmar the rest of the planet uses International Standard.

aanon4, 2024-08-17:
The units shown are selected based on the locale settings in your browser; so no plans to let the admin select what units another user sees.
Mind you, I didnt know Liberia and Myanmar used miles, but the UK certainly does.

kp4djt, 2024-08-17:
Ahh, based on the browser settings. Very well. Yes, I recalled when I was
in the
UK it was sort of a mixed bag... Some things were in Imperial and some in
ISU...
I do recall now that the signs were in miles, but the glass plants I did
work in were
all metric. I recall that weights were metric.

aanon4, 2024-08-17:
We are a backwards and confused people.

closed 2024-08-17 by aanon4

kp4djt, 2024-08-18:
Well, I was born in OKC OK, raised in Puerto Rico, have lived in
various countries
in Latin America, so I am a real units mess. But since I had metric put on
me at an
early age (really had not seized on any system as I was quite young) metric
kind of
took on me. I had been in US Standard long enough that it was in there but
once I
had to start thinking in M, Km, C, etc it became my go to system of units.
Then I
worked in the glass industry for several years, most glass plants are
outside the US
those that are in the US are metric because all of the equipment is metric.

Now in the lab business EVERYTHING is metric, so unless it is an absolute
necessity
most of my measurement units are in metric for my usage. Converted for the
unwashed.

I did enjoy my small amount of time in the UK. Like to go back and visit
again.

Back on the new UI, every time I look at it I find something else that
impresses
me. But I have been setting on asking for something like this  now for some
time.
I had always wondered why you could not see the basic functions of the node
on
the node status landing page. I never got to ask it, and it is MORE than
answered.
Also very impressive for presentations. Sometimes when I show the node
status
page I get this "you mean that is all there is?" look. Now I will no longer
get it.

Thank you again.