aredn/aredn #1289: HAP AC LITE Tunnel Screen reports error - New UI

k1ky, 2024-08-17:
### Describe the bug
A clear and concise description of what the bug is.
Selection of Tunnel Display results in error screen:
ERROR: left-hand side expression is null
[image]

[supportdata-W4CAT-HAP-400-CATHSE-2024-8-17-12-48.tar.gz](https://github.com/user-attachments/files/16645768/supportdata-W4CAT-HAP-400-CATHSE-2024-8-17-12-48.tar.gz)

- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] Yes - N20240817

### Expected behavior
A clear and concise description of what you expected to happen.
Expect Connected Tunnel Stations to be displayed.  I do have another HAP upgraded that works fine.  

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.
[image]

### Additional context
Add any other context about the problem here.

aanon4, 2024-08-17:
Want to confirm that you've not created any tunnels using the new UI, yes?

Looking at /etc/config.mesh/vtun I see various bad tunnel entries. For example, there are lots of tunnels with a netip of '172.31.256.X' when that value should probably be '172.32.0.X'. As 256 is an invalid octet value for an IP address, I think this is why things are breaking. Could you try fixing these by hand and seeing if this resolves the UI issue?

k1ky, 2024-08-17:
Correct - I have NOT added any new tunnels since upgrade.  I'll look at that file and report back.

k1ky, 2024-08-17:
I see the entries with 256 and most are active.  How should I fix this? 
[image]

aanon4, 2024-08-17:
In /etc/config.mesh/vtun, every time you see 172.31.256 you should change it to 172.32.0

k1ky, 2024-08-17:
And I assume I will need to change the client to match?

aanon4, 2024-08-17:
Unknown. They might be fine. The tunnels work after all.

aanon4, 2024-08-18:
Did you manage to clean these tunnels up?

k1ky, 2024-08-18:
Not yet - that's my next project!

k1ky, 2024-08-19:
The node Tunnel Client screen still shows the "old" addys, so there must be another file that needs to be modified? When I "saved" changes in the node, it rewrote the file and eliminated my modifications.  Reopening issue.

aanon4, 2024-08-19:
Need a bit more info here. What "Tunnel Client screen" are you referring to here? The New UI doesn't have such a thing as all tunnels are now under the same UI dialog.

k1ky, 2024-08-19:
Sorry - I meant "Tunnel Server" screen.

aanon4, 2024-08-19:
There isn't a new UI dialog called that; but there is one in the old UI. Do you mean that?

k1ky, 2024-08-19:
Yes, in the old UI Tunnel Server screen - my changes were not reflected.

aanon4, 2024-08-19:
So I'm not going to fix the old UI which it would appear is breaking these things - so if you re-use it, it will re-break them. Is there some reason you need to use the old UI for this still or am I missing something?

aanon4, 2024-08-19:
Let me revisit the old and new code for this. I think there might be a better general solution.

k1ky, 2024-08-19:
Using the old UI to be able to manage the Tunnel Server config info because the new UI fails to display due to the error.  I agree, this is another case of faulty "old data", so I'll probably end up assigning a new starting IP for the HAP Tunnel at some point and move on.

aanon4, 2024-08-19:
I also think the way we are wrapping those addresses is problematic. I told you to increase 172.31 to 172.32 but looking at this further, that'll probably cause different problems eventually. So let me review this some more and get back to you.

aanon4, 2024-08-19:
https://github.com/aredn/aredn/pull/1315

Apparently I lied about not fixing the old code too :)

k1ky, 2024-08-19:
I applied the new update.  The tunnel clients connect to the new 172.32.0.xx addys, but the screen on the old UI still shows the 256 numbers and not reflecting the node connections with a blue cloud even though the client nodes are connected to the new addy. 
[image]

aanon4, 2024-08-19:
Could you try re-editing that vtun file from before, but this time change those 256 values into 0.
I didnt try to fixup the old value, I just tried to not corrupt them again.

aanon4, 2024-08-19:
It's also possible the other end is interpreting an address of 172.31.256.X as 172.32.0.X .. which is what we tried before; but that's problematic as it's a different /16 subnet. So you may need to update the other end - I dont know at this point I'm afraid.

k1ky, 2024-08-19:
I did update the client ends to the new 172.32.0.x  addy.  Pic
[image]
[supportdata-W4CAT-HAP-400-CATHSE-202408182037.tar.gz](https://github.com/user-attachments/files/16654399/supportdata-W4CAT-HAP-400-CATHSE-202408182037.tar.gz)
ture below and will attach new support data file.

aanon4, 2024-08-19:
So unlike the last time I attempted to fix this, the address to update them to is 172.31.0.X and *not* 172.32.0.X

aanon4, 2024-08-21:
Any update on this?

k1ky, 2024-08-21:
Still working on fixing it - gimme another day to review and resolve.  I need to edit the vtun file to 172.0.31.xx addys for the wrap around right?

k1ky, 2024-08-21:
The VTUN Server screen still does not show the new IP assignments as entered into the /etc/config.mesh/vtun file in the old UI..  Is there another one somewhere that also needs to be changed?  New UI shows correct.
[image]
[image]

aanon4, 2024-08-21:
The way the vtun file is fixed is still incorrect. For example, this entry reads:
```
config client 'client_5'
option netip '172.32.0.0'
option enabled '0'
option passwd 'sysman71'
option name 'K1KY-PBELM5XW-149-MQY'
option node 'K1KY-PBELM5XW-149-MQY-172-32-0-0'
option clientip '172.32.0.1'
option serverip '172.32.0.2'
option contact 'vtun file = 172.32.0.0'
```
when it should now read
```
config client 'client_5'
option netip '172.31.0.0'
option enabled '0'
option passwd 'sysman71'
option name 'K1KY-PBELM5XW-149-MQY'
option node 'K1KY-PBELM5XW-149-MQY-172-31-0-0'
option clientip '172.31.0.1'
option serverip '172.31.0.2'
option contact 'vtun file = 172.31.0.0'
```

k1ky, 2024-08-21:
OK  -done and done.  All is well.

closed 2024-08-21 by k1ky