aredn/aredn #1291: New UI: Add clickable link to FW update notice

mathisono, 2024-08-17:
Add clickable link to FW update notice

In UI Status Mode, the FW cell flags with a massage that an update is available. The request is to add a clickable prompt that will lead the admin to the FW tab. Login will be required.

[image]

Orv, 2024-08-17:
That "Update available" flag is itself clickable.

mathisono, 2024-08-17:
Only when in Config mode, Im suggesting that the status mode UI act the same.

aanon4, 2024-08-17:
It does that now if you're authenticated, so do you need something different?

mathisono, 2024-08-17:
This would reduce/ stream lines the clicks need to navigate to the FW tab. At the moment im not suggesting other cell should display a prompt from the Status mode UI, just the FW cell.

aanon4, 2024-08-17:
I dont understand what you're asking for - sorry. If you click that "Update available" banner the firmware tab will open already. I dont understand what UI flow you're asking for which is different to that. Are you suggesting for another message to upgrade somewhere else?

mathisono, 2024-08-17:
The above only works if your logged in.  Im suggesting that if a user is not logged in, "in status mode" the "Update available" would be clickable as well, prompting for a Login and then redirecting to the "FW" tab.

aanon4, 2024-08-17:
That would increase the complexity and size of the status page for logged-out users the majority of whom would never want to click on this. So have to say no here.

closed 2024-08-17 by aanon4