aredn/aredn #1293: NLQ percentage not calculated properly.

Orv, 2024-08-17:
### Describe the bug
NLQ links showing at 500% on certain links.

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes
### Expected behavior
A clear and concise description of what you expected to happen.
Expected 100%
### Screenshots

If applicable, add screenshots to help explain your problem.
[image]

[supportdata-W6BI-SHACK-NODE-2024-8-17-12-7.tar.gz](https://github.com/user-attachments/files/16646266/supportdata-W6BI-SHACK-NODE-2024-8-17-12-7.tar.gz)

aanon4, 2024-08-17:
What are the weights on these tunnels? And are they the same at each end? I assume I've got some math wrong related to tunnel weights.

Orv, 2024-08-17:
Client end: In the old UI, the tunnel weight is set to 5 .  In the new UI, no Wgt values are shown for any tunnels.
Server end: The tunnel weight is set to the default of 1.

aanon4, 2024-08-17:
Is it also 500% if you open the dialog?

Orv, 2024-08-17:
Yes.

closed 2024-08-20 by aanon4