aredn/aredn #1295: Minor visual glitch

Orv, 2024-08-17:
### Describe the bug
A clear and concise description of what the bug is.
A little "scroll arrow" appears on the Time menu, but there's no scroll (see screenshot)
- [ ] Does this issue still occur on the latest version? (**Required**)
Yes
### Expected behavior
A clear and concise description of what you expected to happen.
Shouldn't appear
### Screenshots

If applicable, add screenshots to help explain your problem.

[image]

aanon4, 2024-08-17:
I dont see it on any of my browsers (Safari, Firefox, Chrome on Mac)

Orv, 2024-08-17:
Just noticed it's also at the top.  This is Firefox on Linux.  I'll check Chrome also.

[image]

Orv, 2024-08-17:
I was in error.  Those screenshots are from Chrome on Linux.  Those visual artifacts do NOT appear in Firefox on Linux (Kubuntu 24.04).

k6ccc, 2024-08-18:
This laptop has a fairly low vertical resolution, on the time setting popup, there is a scroll bar with the arrows at top and bottom.  Using Firefox.

aanon4, 2024-08-20:
As this appears to be a browser rendering anomaly, I'm going to hope they fix their browser.

closed 2024-08-20 by aanon4