aredn/aredn #1306: Tunnel Notes not migrating or displayed new UI

k1ky, 2024-08-18:
### Describe the bug
Tunnel Client and Server Connection notes are not transferred/displayed in the new UI

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes - N20240818 

### Expected behavior
A clear and concise description of what you expected to happen.
Expect to see existing notes of Tunnel Clients and Server entries in old UI in the new UI

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

If applicable, add screenshots to help explain your problem.

### Additional context
Add any other context about the problem here.

aanon4, 2024-08-18:
Looks like I just got this completely wrong, not only forgetting to save new notes, but also getting the name of the notes wrong too. Will fix that today.

aanon4, 2024-08-18:
We did an extra build today, so if you want to update your node again and let me know if this is fixed, that'd be lovely.

closed 2024-08-18 by aanon4

k1ky, 2024-08-18:
Yes, that fixed it!