aredn/aredn #1308: Functions broken in nightly build 20240818-f38f753

Orv, 2024-08-18:
### Describe the bug
Reboot function no longer works.

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes
### Expected behavior
A node reboot should occur

### Additional context
I'd add a support data file, but the function to generate one also no longer appears to work.

aanon4, 2024-08-18:
You mean rebooting from the top/right menu? It works when I try it. Do any of the other menu items work/not work?

aanon4, 2024-08-18:
Okay, I can reproduce it one some nodes but not others. Seems likely this is related to changes to allow copy-n-paste to work again. Will look at it.

aanon4, 2024-08-18:
Scrub that. Looks to be crashing the web server on MIPS devices only.

closed 2024-08-18 by aanon4