aredn/aredn #1332: hAP ac lite not displaying firmware status

WT3, 2024-08-20:
MikroTik RouterBOARD 952Ui-5ac2nD (hAP ac lite) not displaying firmware status in new user interface. (IE. update available or up to date.)

[image]

Bill -WB3CQK

aanon4, 2024-08-20:
Give it a bit. It checks every hour if it doesnt know the status, and once a day if it does.

aanon4, 2024-08-20:
Let me know if this status doesnt show up by tomorrow or I will close this bug then - thanks.

WT3, 2024-08-20:
Ok I will keep an eye on it.

Thanks,
Bill

aanon4, 2024-08-20:
This is how it looks to me:
[image]
Looks okay.

WT3, 2024-08-20:
Ok thanks for checking it. I am our of the house at the moment and can't check  from here.

Bill

Screenshot.2024-08-20.at.3.32.30.PM.png (view on web)<https://github.com/user-attachments/assets/dac774f3-1275-4195-9ef9-48d030266615>

closed 2024-08-21 by aanon4