aredn/aredn #1337: Display difference between old and new UI

Orv, 2024-08-21:
### Describe the bug
In the old UI, the DHCP configuration shows the starting and ending IP addresses in the pool.
In the new UI, it appears to show the starting and ending ACTIVE (issued) addresses.  The old version would be preferable in my opinion

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes
### Expected behavior
Same behavior between old and new UIs

### Screenshots
[image]

[image]

aanon4, 2024-08-21:
You're on today's nightly, yes? I dont see this today. What's the node name so I can look?

aanon4, 2024-08-21:
Worth noting that if you change the DHCP size value from 5 to 13 *before* todays update, these values will be wrong. Best to change it again.

aanon4, 2024-08-21:
Looks like it was set to 13 when this feature was broken. Now that's correct. the fix was to change it to 29 and then back to 13 again and commit.

closed 2024-08-21 by aanon4