aredn/aredn #1343: Show firmware update pop-up in landscape mode.

nc8q, 2024-08-21:
**Is your feature request related to a problem? Please describe.**
The firmware pop-up page need to be scrolled vertically on my laptop screen.

**Describe the solution you'd like**
Please organize the pop-up screen in landscape instead of portrait so that it can be displayed without vertical scrolling.

**Describe alternatives you've considered**
Mobile phone users can rotate their phones 90 degrees into landscape to view the firmware pop-up.
**Additional context**
I assume that when folks are updating their firmware, they do so from a device with a 4:3 or 16:9 display.

aanon4, 2024-08-21:
I think the mobile layout will need to work in portrait, even if it can also work in landscape.
To address this, I've compress the firmware+hardware information. Let me know how this looks on your laptop once this is in the build.

closed 2024-08-22 by aanon4

nc8q, 2024-08-26:
Hi, Tim:
I tested on 2 laptops.
Refresh now appears.
Advanced Settings are off screen...need to scroll to access.
Good enough.
73, Chuck