aredn/aredn #1345: DHCP server not allocation addresses correctlly

Orv, 2024-08-21:
### Describe the bug
After switching DHCP server from host 5 to host 13 (and IP addresses being rearranged as expected), the DHCP server still refuses to hand out more than 5 IP addresses.
- [ ] Does this issue still occur on the latest version? (**Required**)
Yes
### Expected behavior
All other devices attached to hAP ac2 (at least 2, via switch) should receive IP addresses in allocated range
[supportdata-W6BI-SHACK-NODE-2024-8-21-10-32.tar.gz](https://github.com/user-attachments/files/16695882/supportdata-W6BI-SHACK-NODE-2024-8-21-10-32.tar.gz)

aanon4, 2024-08-21:
Looks like I missed one of the values (too much redundancy in the dhcp values - I really need to fix this at some point).

aanon4, 2024-08-22:
Note. Once you have update your node's firmware, go into the Network settings and set the DHCP range to be higher than you want; then set it lower than you want (no need to leave the dialog between these two steps). Then OK then Commit.

Orv, 2024-08-22:
Now working as expected - thanks!

closed 2024-08-22 by aanon4