aredn/aredn #1360: Ubiquiti PBE-5AC-GEN2 Firmware Install Failure

dan97520a, 2024-08-22:
### Describe the bug
Newer PowerBeam PBE-5AC-GEN2 can only downgrade to 8.7.4 so AREDN 3.24.6.0 factory firmware fails install with:  Invalid version 'WA.ar934x.v8.5.3-42.AREDN-r23809-234f1a2efa' 

Related bugs: #1262 and #1353

- [Yes ] Does this issue still occur on the latest version? (**Required**)

### Expected behavior
Expected AREDN firmware to load and boot.

### Screenshots
Please include a screenshot of your system information (if possible) if the specific system environment is relevant to the bug.

### Additional context
scp aredn-3.24.6.0-ath79-generic-ubnt_powerbeam-5ac-gen2-squashfs-factory.bin admin@192.168.1.20:/tmp/factory.bin
aredn-3.24.6.0-ath79-generic-ubnt_powerbeam-5ac-gen2-squashfs-factory.bin           100% 5697KB   1.6MB/s   00:03

C:\>ssh admin@192.168.1.20
 ___  ___
|\  \|\  \      Welcome to airOS
\ \  \\\  \
 \ \  \\\  \    PRODUCT: PowerBeam 5AC
  \ \  \\\  \   MAC: E43883ACC6C1
   \ \_______\
    \|_______|  VERSION: WA.v8.7.4.45112

WA# hexdump -Cv /bin/ubntbox | sed 's/14 40 fe 27/00 00 00 00/g' | hexdump -R > /tmp/fwupdate.real
WA# chmod +x /tmp/fwupdate.real
WA# /tmp/fwupdate.real -m /tmp/factory.bin

Invalid version 'WA.ar934x.v8.5.3-42.AREDN-r23809-234f1a2efa'

aanon4, 2024-08-22:
Waiting on feedback from https://github.com/aredn/aredn/pull/1353 before making the same change for this device.

aanon4, 2024-08-23:
Please verify in tomorrow's nightly.

dan97520a, 2024-08-23:
Progress, but firmware will not boot …

aredn-20240823-c7d942a-ath79-generic-ubnt_powerbeam-5ac-gen2-squashfs-factory.bin     100% 5761KB   2.1MB/s   00:02

WA# /tmp/fwupdate.real -m /tmp/factory.bin
Writing 'kernel         ' to /dev/mtd2(kernel         ) ...  [%100]
Writing 'rootfs         ' to /dev/mtd3(rootfs         ) ...  [%100]
Done
Pinging 192.168.1.1 with 32 bytes of data:
Reply from 192.168.1.10: Destination host unreachable.
Reloaded 8.7.4 and tried again, same result - bricked.

dan97520a, 2024-08-23:
Accidentally closed, so reopened.

aanon4, 2024-08-23:
Not much I can do at this point without access to failing hardware. If it wont boot onto the network then I'd have to get one with the known problem, destroy the case to get to the serial interface, and debug it from there. And that might not get me anywhere.

When you boot the "bricked" device, does it produce any network traffic at all (captured using tcpdump or wireshark)?

dan97520a, 2024-08-23:
No packets from the device after firmware install.  I will install the firmware again and see if the device has anything to say at the end of the firmware load.  A little disappointing, I bought a 5-pack of these because they are fully supported in the AREDN hardware reference.

aanon4, 2024-08-23:
Ubiquiti updates it hardware without updating the versioning. Mikrotik too of course. I have PBE-5AC-GEN2 stuff that works, but didnt have the downgrade limitation. This is the first device I'm aware of which has both that limitation, and once we worked round it, still doesnt work.

Assuming re-attempting this doesnt resolve the issue, put the Ubiquiti software back on this device and then - I hope - you can return them

dan97520a, 2024-08-23:
Successfully installed NanoBeam AC Gen2
New UI reports PowerBeam 5AC 400
Seems to be working normally.

aanon4, 2024-08-23:
Could you attached the support data so I can take a look at why its misreporting the device?

dan97520a, 2024-08-24:
[supportdata-K7CIJ-Test-1-2024-3-22-22-17.tar.gz](https://github.com/user-attachments/files/16734822/supportdata-K7CIJ-Test-1-2024-3-22-22-17.tar.gz)

aanon4, 2024-08-24:
Slightly confused on re-reading so I need a little confirmation before committing more changes.
Did you install the nanobeam firmware on a powerbeam to get it to work because the powerbeam firmware did not?

aanon4, 2024-08-24:
Also, it appears the PowerBeam 5AC 400 and the PowerBeam 5AC Gen2 are the same thing, just with different names. The Ubiquiti specs look the same and the lowest level hardware configuration also looks the same.

dan97520a, 2024-08-24:
Yes, the PBE-5AC-Gen2 is a 400mm dish product. The PBE-M5-400 is also a 400mm dish with the AC gigabit spec. I could not find a product called PowerBeam 5AC 400.

dan97520a, 2024-08-24:
Looks like I may have installed the powerbeam ac 400 firmware but I wrote what worked in the last comment.  I'll install again in a few hours from now and make sure. Dan, mobile.

dan97520a, 2024-08-24:
20240823 Nightly build installation summary:

powerbeam-5ac-400    Invalid
powerbeam-5ac-gen2  Installs and bricks

nanobeam-ac-gen2      Installs and works
nanobeam-ac-gen2-xc Invalid

litebeam-ac-gen2         Installs and bricks

aanon4, 2024-08-25:
PowerBeam 5AC 400 - https://dl.ubnt.com/datasheets/PowerBeam_ac/PowerBeam5ac_DS.pdf - but it looks like the renamed it at some point. I'll update its display name in AREDN to note this.

https://github.com/aredn/aredn/pull/1392

Thanks for the information. For now I suppose you can go with what works. Odd though, as the powerbeam-5ac-gen2 is a standard OpenWRT base image, so I'd expect it to work without any changes from AREDN.

The nanobeam is a dual ethernet product; does it show up as having that feature?

jimfenton, 2024-09-23:
For what it's worth, I recently (August) bought a PBE5AC-GEN2 and flashed it with one of the daily builds successfully and without incident. Currently running 20240906-f5ef0ec.

/K1JH

dan97520a, 2024-09-23:
I just did a clean install of airos 8.7.4 firmware and then aredn 20240923-39981bc factory pbe-5ac-gen2 firmware and received an “Invalid version ‘WA-ar934x.v8.5.3-42.AREDN-r23809-234f1a2efa’ error.

 

Dan, K7CIJ

aanon4, 2024-12-04:
Closing. Assuming this is also the problem seen here https://github.com/aredn/aredn/issues/1672, so tracking this there.

closed 2024-12-04 by aanon4