aredn/aredn #1363: firmware slider bar obscures portion of refresh icon

nc8q, 2024-08-22:
### Describe the bug
Aiming at the firmware refresh icon is awkward because the icon is partially obscured by the scroll slider bar.
- [ ] Does this issue still occur on the latest version? (**Required**)
- [ ] Yes. Nightly 20240822

### Expected behavior
Easy access to refresh icon or
no vertical scrolling required.

[image]

aanon4, 2024-08-22:
The problem here is I have no idea if there is a scrollbar, how wide it is, when it appears, or if it changes size when you scroll. In your setup do you see it all the time, or only when you start to scroll?

nc8q, 2024-08-22:
Scroll-bar appears whenever I hover my mouse pointer over the pop-up page.
My scroll-bar gets larger when I hover over the right side of the pop-up page
near the Help, text boxes, refresh icon, and Done.
What if, instead of a teeny tiny icon on the right-margin, the left text
"Download Firmware
Download firmware from an AREDN server."
was a 'hot-link'?
The 'title' is 'firmware configuration', so maybe
the text 'Refresh' of the left side, as a hot-ink, would suffice.
OBTW, on my mobile device that entire page loads in portrait
which makes everything extremely teeny tiny.
[image]

aanon4, 2024-08-22:
This is not what the mobile layout will look like, so best not to use that for any comparison purposes.

closed 2024-08-23 by aanon4