aredn/aredn #1364: Firmware config Window has two scroll bars

K5RA, 2024-08-22:
Firmware config screen has two scroll bars.  Neither active until Advanced options is activated.  When inboard scroll is active, outboard still asleep.  Both scroll bars are in the window whether cursor is there or not.   
Inboard scroll bar should be slightly further to right.  Second scroll bar not expected.
System is Mikrotik hAP ac with 20240822 nightly, rp5b with latest bookworm

[image]

aanon4, 2024-08-22:
Could you provide your platform and browser information - specific version numbers for each - thanks.

K5RA, 2024-08-22:
RB952Ui-5acnD (hAP ac lite)
Raspberry Pi 5B rev 1.0, Linux 6.6.31+rpt-rp1-2712, Debian 12.2.0-14, Debian GNU/Linux 12
Chromium 126.0.6478.164

aanon4, 2024-08-22:
I can fix the double scrollbar, but I'd like to know if this persist on a newer version of Chromium on the Pi; assuming 127.X is available?

K5RA, 2024-08-22:
I am running updates on my r-pi now.  Will have info in about ten minutes.

K5RA, 2024-08-22:
I ran our maintenance scripts with   sudo apt -y update   and   sudo apt full-upgrade  .  Chromium version did not change.  Give me a bit and I'll try sudo apt-get install chromium-browser

K5RA, 2024-08-22:
Well...,that was easy.  It said what I have is the a;ready the newest version.

closed 2024-08-23 by aanon4