aredn/aredn #1365: Favicon needed!

Orv, 2024-08-22:
Would be helpful to have a distinctive favicon for AREDN node web pages.
**Describe the solution you'd like**
Would be cool to have a mini version of the AREDN logo as a favicon, if possible.

aanon4, 2024-08-22:
Does anyone have a favorite online favicon maker?

VA2XJM, 2024-08-22:
https://realfavicongenerator.net/

Jean-Michel Vien - VA2XJM
Président, Association des Radioamateurs de l'Abitibi-Témiscamingue
https://www.ve2atu.org

dman776, 2024-08-22:
favicon15 KBThanks,Darryl

aanon4, 2024-08-22:
Thanks.

Looks like Firefox and Chrome support SVG favicons so we will go with that. Safari doesn't (my preferred browser) but I'm not carrying and loading an extra icon resources just for a browser which is 3 years behind everyone else on this. Hardly critical functionality.

Similarly, the favicon will be loaded from the localnode regardless of which node you're looking at on the network. This is how all resources work (it saves time and bandwidth) but I wont bother falling back to a network load because a nice icon is not critical.

aanon4, 2024-08-23:
Checked in Firefox and Chrome (wont work on Safari because .. Apple)

closed 2024-08-23 by aanon4