aredn/aredn #1370: Local nodes now listed under Neighbor Nodes

Orv, 2024-08-22:
### Describe the bug
Local Nodes list is now empty;  Local Nodes now shown under Neighbor Nodes

- [ ] Does this issue still occur on the latest version? (**Required**)
Yes
### Expected behavior
A clear and concise description of what you expected to happen.
Nodes sorted as before

### Screenshots
[image]

[supportdata-W6BI-SimiNorth-5G-2024-8-22-16-47.tar.gz](https://github.com/user-attachments/files/16720223/supportdata-W6BI-SimiNorth-5G-2024-8-22-16-47.tar.gz)

aanon4, 2024-08-22:
Only DtD nodes which are close are considered local - everything else is a neighbor.

These mostly look to be wifi connected, except for one which I assume has a sufficiently different lat/lon.

aanon4, 2024-08-23:
I checked the two DtD connected nodes, the both at > 50 meters from this node, so are not considered local. This information looks correct to me.

Orv, 2024-08-23:
I'll adjust the lat/lons as appropriate and recheck.

Orv, 2024-08-23:
>50 meters was done so nodes aren't stacked on top of each other on the map.  Maybe 100 meters would be more appropriate?

aanon4, 2024-08-23:
> > 50 meters was done so nodes aren't stacked on top of each other on the map.  Maybe 100 meters would be more appropriate?

Did I say that? To be honest I'd rather the distance was smaller. If two local nodes are 49 meters apart, that's a really long way for something to be considered at the same place (but we do).

Orv, 2024-08-23:
I'd say the Ethernet cable limit should be used: 100 meters.  :-)

Orv, 2024-08-23:
Whatever it ends up being, it should be called out in the docs (if it's not already).

aanon4, 2024-08-23:
@ab7pa Let's call out the specifics of what a 'local' node is in the docs.

closed 2024-08-23 by aanon4